randdusing/cordova-plugin-bluetoothle #219: Android Crash on Initialisation / Scanning

maximilianloy, 2015-12-11:
Hi,

Once in a while I see a crash of the app as soon as it initialises / starts scanning.
I can not reproduce it but would like to post the crash log here in case it makes sense to someone.

My device is a Nexus 5 running Android 6.0

[bt-crash-log.txt](https://github.com/randdusing/BluetoothLE/files/59710/bt-crash-log.txt)

randdusing, 2015-12-23:
I'm not sure what the BluetoothPairingDialog is about, but I believe the plugin related errors are from a race condition. Still trying to figure that one out...

randdusing, 2016-01-16:
Still seeing this issue? I took another look at your crash log and it looks to be related to an RSSI call. Are you calling that anywhere?

randdusing, 2016-01-25:
I believe this should be fixed in v3 with some changes to the scan locking mechanism. Let me know if it still exists.

closed 2016-01-25 by randdusing