randdusing/cordova-plugin-bluetoothle #222: IOS seems to stop receiving after fixed time

diverJohn, 2016-01-01:
We have been using your BT plugin, v1.0.6, and currently updating to the latest in npm, v2.7.0.
We are using Phonegap Build and IOS phone is at version 8.1.3.
BT Tx is WRITE-NO-RESPONSE
BT Rx is subscribed.

The problem occurs on IOS only when there is a lot of communication between our App and our device via BTLE.   It seems that the app somehow turns BTLE reception off after about 5 minutes and the app times out.  Note that the same application communicating to the same hardware works fine on Android.  If I dismiss the app I can see several BTLE messages come in, via console, as if the app wakes up to be dismissed.  I also use the powerManagement plugin to disable AutoLock.

The problem occurs with v1.0.6 of the plugin, which used to work with earlier versions of IOS.  The problem also occurs with npm version 2.7.0.   Hopefully you or someone is familiare with this issue and can suggest a solution.

randdusing, 2016-01-01:
I'm not sure if I understand your last paragraph. Have you always experienced this issue, or was it recently introduced with a new plugin version?

And can you give a little more detail on what you're doing? I may be able to try recreating it on my end.

diverJohn, 2016-01-01:
This is a recent issue that occurs on the old BT plugin, v1.0.6, and the latest plugin, v2.7.0.   We released our app using v1.0.6 about 1 year ago and everything worked.   For the last few months we have been adding capability to our app, newer plugins, newer Phonegap etc., but keeping the BT plugin the same at version 1.0.6.   When we started more rigorous testing we noticed that the BT communication stalled on IOS after several minutes.   I thought that there might be an issue using the old BT plugin with a more recent Phonegap Build so I updated the project to the latest BT plugin but still had the problem on IOS.

Detail:
We transmit anywhere from 12 to 250 bytes at a time to our device from the phone.  The protocol is expects a response from our device before transmitting the next message.   If no response is received in 1 second we try again for several times before giving up.
Let me know if this is enough for you to go on or if I need to send you anything else.

randdusing, 2016-01-02:
Ok, currently I don't have a good way to test large data transmissions, but I should within a week or so.

You may want to add some NSLog statements to the iOS code in the didUpdateValueForCharacteristic or didWriteValueForCharacteristic methods and see if those events are firing. If you still see those events firing, then something must be wrong with calling the callback.

diverJohn, 2016-01-03:
Since I'm using Phonegap Build I don't think that I have access to "didUpdate/WriteValueForCharacteristic" methods.

When I revert back to Phonegap 3.6.3 using Phonegap Build I no longer have the problem.   It looks like the problem started occurring when I made the change from Phonegap 3.6.3 to 3.7.0.  

Do you know if Phonegap 3.7.0 allows the BT to go into some power saving mode in IOS 8?

randdusing, 2016-01-03:
Ahh, I forgot you were using PhoneGap Build. Are you able to test it on newer version of PhoneGap Build? There were no changes like that in iOS 8 that I'm aware of

diverJohn, 2016-01-04:
When I try the latest PhoneGap in my Phonegap Build I get a couple of file and file transfer plug-in conflicts.  I may have to try to sort those out.

diverJohn, 2016-01-04:
I updated Phonegap to the latest available in Phonegap Build, 5.2.0, and the problem was resolved.   You can close the issue.    Note that we are still using v1.0.6 of your BTLE plugin.

Upgrading to Phonegap 5.2.0 caused other plugin issues.  
I had to upgrade two other plugins, file and file-transfer, that we were using.   I also had to include a new plugin for network connectivity, "whitelist".

        
    <gap:plugin name="cordova-plugin-file"           source="npm" version="3.0.0" />
    <gap:plugin name="cordova-plugin-file-transfer"  source="npm" version="1.4.0" />        
    <gap:plugin name="cordova-plugin-whitelist"      source="npm" version="1.2.0" />

randdusing, 2016-01-05:
Cool, glad you got it working, but the error is a bit concerning. I'll keep this open, so I remember to test out data transmission when the server role code is completed.