randdusing/cordova-plugin-bluetoothle #223: NullPointerException

jkhax0r, 2016-01-06:
I'm getting the following error / crash reports.  This is on 2.3.0.  

My app only scans.  No connections.  For whatever reason (I think it was weird Nexus 4 behavior), I stop and restart scanning every 5 seconds or so.  

This crash seems to come much more frequently when I have a lot of devices and speed up advertising of them.  I was tempted to just put some checks for NULL and see if it goes away but it looks like the code already does what I was going in to try. 

This is a bit difficult to reproduce but I think if I take 20 or so of my devices and put all of them advertising very quickly I could reproduce it in a few minutes.  

Thanks in advance for any help or ideas you have.  Hopefully this image scales well enough to read...

[image]

randdusing, 2016-01-07:
I've noticed some similar issues before. I believe it's an issue with multithreading and not keep a lock on the callback. I'm hoping to make some large updates in the near future, and adding proper synchronization should fix these types of issues.

randdusing, 2016-01-16:
How many devices are advertising? I can't recreate it, maybe because I don't have enough devices, but I think the scan result and stop scan code are running at the same time. If you just turn on scan once without going back and forth, can you recreate the issue? If my hypothesis is right, you shouldn't be able to recreate it. Either way, I've updated the code in my working version that should fix the issue.

jkhax0r, 2016-01-19:
Well, I just tested the opposite...  I sped up the code that stops and restarts scans to do it every 200-300ms (random).  Then with 3 devices advertising at ~200ms each, I was able to cause the crash 3 times in a row in less than a minute or two.

As a side and hopefully quick question, I saw that the latest version requires SDK=23.  Does this mean that the plugin won't run on older phones that don't use the latest android?

randdusing, 2016-01-19:
Oh that makes sense. Let me see if I can recreate it with a very small scan interval. I originally added this to my next major version, but I'll see if I can add this to unrelated minor update I should be releasing today.

It will build against SDK=23 but still runs with older versions of Android. Some of the functionality will just be limited since it wasn't available in prior versions.

jkhax0r, 2016-01-19:
Great thank you!  Let me know if there is anything I can do to help.  

As a side note, if you provide me a shipping address I'd be happy to ship you a few of my devices.  Just in appreciation for your efforts.  They are transmit only and send an advertisement every 4 seconds until you press the button, in which case they transmit at 200ms interval for the next 20 minutes.  I can send you as many as you want for either personal use or as test devices to fill the airwaves.

https://play.google.com/store/apps/details?id=com.mopeka.tankcheck
https://itunes.apple.com/us/app/mopeka-tankcheck/id981959807?mt=8

randdusing, 2016-01-19:
Hmm, does this only happen when they begin transmitting every 200ms? I have a decent amount of devices, but they don't advertise nearly as often. I tried switching the scan on and off every 50ms and haven't ran into any crashes yet. This is on a Nexus 7 running Android 6.0.1. Either way, the changes are pretty minor to make, so I can see if that solves the issue.

My email is {{github username}}@gmail.com, want to send me an email? The devices sounds pretty cool and should help debug the issue.

jkhax0r, 2016-01-19:
Okay.  I will try it out asap and let you know. Let me know when its committed.

randdusing, 2016-01-19:
Try out the scanlock branch, should be able to install it with:

```cordova plugin add https://github.com/randdusing/BluetoothLE.git#scanlock```

randdusing, 2016-01-25:
Added this to the latest version, v3. Let me know if there's still an issue.

closed 2016-01-25 by randdusing

jkhax0r, 2016-01-25:
Thanks!  Sorry I haven't been able to test yet.  I use phonegap build and it looks like they don't have android-cordova >= 5.0.0 yet so I'm going to have to go back and build on the desktop.  Should be sometime this week.

randdusing, 2016-01-25:
Ahh ok, I wish PhoneGap Build updated quicker...