randdusing/cordova-plugin-bluetoothle #225: No way to attach callbacks to peripherals returned by retrieveConnected

additv, 2016-01-14:
At the moment it seems like there is no way to attach the connectSuccess and connectError callbacks (that you would normally create in a call to connect()) to an already connected peripheral returned by retrieveConnected.  Calling reconnect with the peripheral address results in an isNotDisconnected error.

The only way I can see is calling disconnect on the peripheral, and then creating a new connection.  The problem with this is that certain peripherals go to sleep after a disconnect rather than begin advertising again.

The main problem with not being able to connect these callbacks is we then miss out on disconnection notifications (which go through connectSuccess).

randdusing, 2016-01-14:
Documentation should be more clear. Retrieve connected returns "paired" devices rather than devices the app is actually connected. RetrieveConnected is from iOSs function name. Here's some more info: https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/#//apple_ref/occ/instm/CBCentralManager/retrieveConnectedPeripheralsWithServices:

closed 2016-01-14 by randdusing

additv, 2016-01-14:
Thanks for the reply!  I understand what retrieveConnected does, so let me clarify.
Is there a way to attach a disconnect listener to these already paired devices returned by retrieveConnected?

Sometimes I need to reconnect to an already paired device.  These devices don't show up in a scan because they are connected, but I can't connect/reconnect to them because they are "not disconnected".  They're in a third state, which is connected but without the connect callbacks.

I can go ahead and use them (RSSI, and subscribe work just fine), but if they disconnect I can't get a notification because they weren't connected through connect/reconnect.  

Does that make sense?

randdusing, 2016-01-14:
Not sure if I follow.. You can't attach a listener to them since a connection has never been initiated by the app. Can you post some code snippets or a summary of what calls you're making?

additv, 2016-01-14:
Sure, here's a general use case and then my specific situation:

In the general case, it seems like iOS allows for multiple applications to connect to a single peripheral (while one app or the other is backgrounded).  A peripheral that has already been connected to by another app, won't show up in a scan, but can be used by getting its address from retrieveConnected.

My specific situation is slightly different, but similar enough:
I'm developing an app using Ionic, which in turn uses Cordova.  When running an application through the CLI it supports a live-reload functionality which refreshes the javascript when you save an edit.  Of course, if you trigger a reload after you connect to a peripheral iOS stays connected to the peripheral. This means it won't show up in scans, and the only way to reacquire it is through retrieveConnected, but then you can't attach a callback to watch for disconnects.

My basic api flow in this case is:
initialize()
retrieveConnected()
subscribe() /read()/write()/RSSI()/etc.  

When you get the UUID from retrieveConnected, it seems like you can just use it in the same way as if you got it through scan/connect.  You just don't get the disconnect callback. 

I'm sure there are other situations where this might become an issue, but I would have to do more research into the nuances of retrieveConnectedPeripheralsWithServices: to be sure.

randdusing, 2016-01-14:
Ahh I see what you're doing. I ran into a similar issue when working with live reloading as well and got around it by closing all previously connected UUIDs on the refresh.Then I went through the normal scan, connect, discover, subscribe/read/etc process. I don't think this state would occur outside of live refresh debug situations, so I don't think it makes sense to update the plugin to handle it.

vmandy, 2018-07-31:
Hi @randdusing, great plugin!

Actually, aside from the live reload case, there is production scenario where this could be useful.
As @additv mentioned quickly above, what if I would like to share/use a BLE connection to a peripheral from 2 separate apps? 

Could  retrieveConnected method be used to re-obtain the currently connected peripheral's address and name? Would there then be a way to update the state of the plugin to consider itself connected to that peripheral (something like a 'reattachTo(device)' method). If possible, then that second app could reuse an existing BLE connection.

Is there any way to accomplish this currently?

Funkelfetisch, 2021-08-29:
> Hi @randdusing, great plugin!
>
> Actually, aside from the live reload case, there is production scenario where this could be useful.
> As @additv mentioned quickly above, what if I would like to [...]

+1