randdusing/cordova-plugin-bluetoothle #226: indications support

jouioui, 2016-01-15:
Hello,
I'm trying to get indications from a ble device, I correclty get notifications but no indication.
I'm wondering if it is a pb from my ble device or not,

From what I see in the limitations of the plugin, it seems supported but not tested.
Can you confirm it is not tested ?

thanks and regards

randdusing, 2016-01-16:
I haven't tested indications, so it's possible there's an issue with the code. Do you know of any cheap devices supporting indication, I could then test it pretty easily.

jouioui, 2016-01-18:
We use our own BTLE custom device, so I haven't tested on an existing one that supports indications.
 
As we need an ACK to forget the data on the BLE device, a workaround is to implement a fake write on the characteristic. We'll try that

thanks

closed 2016-01-18 by jouioui