randdusing/cordova-plugin-bluetoothle #228: Data transfer

LNTWave, 2016-01-18:
Hi,

When I try to send a junk of data (>20byte) to the plugin, the data received at the receiving end is not complete. It is always equal to 20byte. 
On the other hand, if I try to alter your plugin code, such a way it splits the data into 20bytes it

randdusing, 2016-01-24:
Hi @LNTWave, I would try splitting the data into 20 byte packets from the app level.

OrenLederman, 2016-03-23:
The 20 bytes limit is because of the Bluetooth LE protocol. I think it's increased in Bluetooth 4.2, but it doesn't seem to be supported by all phones.

randdusing, 2016-03-23:
@LNTWave Check out the writeQ functions that I added recently with 3.2.0. This will split the packets and write them as fast as possible on Android and iOS. You'll need to install directly form GitHub since it's not in NPM yet.

closed 2016-03-23 by randdusing