randdusing/cordova-plugin-bluetoothle #230: Add Android support for BluetoothManager.getConnectedDevices()

maximilianloy, 2016-01-21:
Hi,

We need to be able to get a list of connected devices as provided by Android with BluetoothManager.getConnectedDevices(). The retrieveConnected() function of the plugin is a bit misleading as it actually returns a list of bonded/paired devices, but they might not be connected.

I think it would make sense that:

* retrieveConnected() calls bluetoothManager.getConnectedDevices() internally
* introduce a new function called retrieveBonded() which calls bluetoothAdapter.getBondedDevices() internally

This is critical for our app so let me know how I can help to make it happen!

randdusing, 2016-01-21:
The naming definitely needs improvement. iOS uses retrieveConnected for "paired" devices, so the terminology made it's way into the plugin.

GetConnectedDevices() will return devices that may not necessarily be connected to your app, will that cause any issues? I think that's why I originally decided not to implement it since the connected devices could be tracked by the apps logic.

maximilianloy, 2016-01-21:
No, we would filter out the relevant devices by name or service UUIDs.

The issue we see is that the app could have been closed without prior disconnect (our app relies on staying connected) and on some occasions the connection stays up, but a new instance of the app is created when started again.

randdusing, 2016-01-21:
The issue I see is getting the app to "connect" back to the devices. Right now I'm not sure if that's possible without making some major changes to the plugin. The GetConnectedDevices() function itself is pretty simple to implement. But right now I'm strapped for time, so not sure how soon I can get it added.