randdusing/cordova-plugin-bluetoothle #231: Permissions issue on Android with min-sdk set to 23 as per android quirks note

msudol, 2016-01-22:
This just started happening out of the blue within the last few weeks. I'm going to try and test this on some other android devices that have not been updated to the latest version as of Jan 10, 2016 to see if this was introduced by Android.

Here is my android monitor log:

```
01-21 15:52:12.588: I/chromium(16571): [INFO:CONSOLE(415)] "startScanning: Serive UUID:26080ee3-2174-4f8f-a65e-93094637dc1c", source: file:///android_asset/www/demo_js/current_index_ble.js (415)
01-21 15:52:12.590: D/BluetoothAdapter(16571): startLeScan(): null
01-21 15:52:12.594: D/BluetoothAdapter(16571): STATE_ON
01-21 15:52:12.595: D/cr_Ime(16571): [ImeAdapter.java:213] updateKeyboardVisibility: type [0], flags [0], show [true]
01-21 15:52:12.596: D/cr_Ime(16571): [AdapterInputConnection.java:178] updateState [] [0 0] [-1 -1] [true]
01-21 15:52:12.598: D/BtGatt.GattService(12857): registerClient() - UUID=8bb6ac8b-77d9-4774-8a48-c07d976ecae0
01-21 15:52:12.598: D/BtGatt.GattService(12857): onClientRegistered() - UUID=8bb6ac8b-77d9-4774-8a48-c07d976ecae0, clientIf=5
01-21 15:52:12.598: D/BluetoothLeScanner(16571): onClientRegistered() - status=0 clientIf=5
01-21 15:52:12.598: D/BtGatt.GattService(12857): start scan with filters
01-21 15:52:12.603: W/Binder(16571): Caught a RuntimeException from the binder stub implementation.
01-21 15:52:12.603: W/Binder(16571): java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
01-21 15:52:12.603: W/Binder(16571): 	at android.os.Parcel.readException(Parcel.java:1620)
01-21 15:52:12.603: W/Binder(16571): 	at android.os.Parcel.readException(Parcel.java:1573)
01-21 15:52:12.603: W/Binder(16571): 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:772)
01-21 15:52:12.603: W/Binder(16571): 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:324)
01-21 15:52:12.603: W/Binder(16571): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
01-21 15:52:12.603: W/Binder(16571): 	at android.os.Binder.execTransact(Binder.java:453)
01-21 15:52:14.603: I/chromium(16571): [INFO:CONSOLE(823)] "scanSuccess", source: file:///android_asset/www/demo_js/current_index_ble.js (823)
01-21 15:52:14.610: I/chromium(16571): [INFO:CONSOLE(475)] "handleBLECallback", source: file:///android_asset/www/demo_js/current_index_ble.js (475)
01-21 15:52:14.611: I/chromium(16571): [INFO:CONSOLE(477)] "{"status":"scanStarted"}", source: file:///android_asset/www/demo_js/current_index_ble.js (477)
```
and in my androidmanifest.xml

    <uses-sdk android:maxSdkVersion="23" android:minSdkVersion="23" android:targetSdkVersion="23" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

randdusing, 2016-01-22:
Starting with API 23, you need to request the permission even if it's included in the manifest.xml. You can use the hasPermission and requestPermission methods to request it at runtime.

msudol, 2016-01-22:
Pardon me for being a neophyte, can you point me in the direction of how to do that?

randdusing, 2016-01-22:
Sure, https://github.com/randdusing/BluetoothLE#haspermission

msudol, 2016-01-22:
oh my, RTFM - I must have looked right past that.. thank you very much.

msudol, 2016-01-22:
Does this have to be done at the time of every scan or can I just do it when the app initialized?

randdusing, 2016-01-22:
Usually just one per app install. But I would check permissions each time
the app is run to see if the user disabled the permission for some reason

closed 2016-01-22 by msudol