randdusing/cordova-plugin-bluetoothle #234: Initialize callback functions are inconsistent

timdelawe, 2016-02-09:
I've wondered why the callback in initialize doesn't work. Solved on stackoverflow: [callback issue](http://stackoverflow.com/questions/35289499/named-callback-function-doesnt-work/35289778?noredirect=1#comment58290172_35289778)

randdusing, 2016-02-09:
Hi Tim, initialize only has the success callback with the last version of the plugin. It should watch for status => enabled or status => disabled events. The documentation does need an update.

timdelawe, 2016-02-10:
Okay, thanks for your answer. But I still have the probleme that the callback function isn't called:

`onInitialized: function(iniObj)
                    {
                        if (iniObj.status === "enabled")
                        {
                            document.getElementById("testvalue").value += "\ninitialized";`
                        }
`bluetoothle.initialize(appEvent.onInitialized, {"request": true, "statusReceiver": false});`

randdusing, 2016-02-10:
StatusReceiver should be true or you won't receive the status updates.

timdelawe, 2016-02-11:
Yes I've allready tried but problem is still the same. I've moved things around a few times but don't get it run. 
With version 2.0.0 and it's 2 callback functions it works great. 

Here is my whole code:
[initialization_issue.txt](https://github.com/randdusing/cordova-plugin-bluetoothle/files/126011/initialization_issue.txt)

randdusing, 2016-02-12:
Can you try using the example app with the Angular wrapper and see if that works? If it doesn't there may be an issue with the device. If it works, there's most likely an issue with your app that we aren't seeing.

timdelawe, 2016-02-19:
I have tried a few ways but I can't commands like `bower install git://github.com/randdusing/ng-cordova-bluetoothle.git#master` in Adobe Build. The only additional informations I can give is that I run the app on a Nexus 6 with android 5.1.1.

randdusing, 2016-02-19:
Hmm, I'm not sure how Adobe Build handles bower. You may just need to edit the bower.json file

timdelawe, 2016-02-22:
Ok it was a bad mistake of mine. I have not seen that Adobe Build still uses `cordova-android 4.1.1`.
I've started using phonegap on ubuntu now.
Thanks a lot for your patience.

closed 2016-02-22 by timdelawe

randdusing, 2016-02-22:
No worries, glad you got it working.