randdusing/cordova-plugin-bluetoothle #236: RSSI 0 - question.

spiritonline, 2016-02-14:
Hi there,

I'm just starting out with programming but I've used your awesome plugin to check if a connected BLE device is disconnecting. On my nexus 6P, as soon as I turn off the bluetooth device cordovaBluetoothLE.rssi returns 0. I'm using this 0 to detect a disconnect much quicker then waiting for the actual disconnect. 

However, on any other android device rssi will stay equal to the last value before disconnecting. 
Could you please help me understand what's happening?

Cheers,
Jason.

randdusing, 2016-02-14:
Hi Jason, I never thought of trying a disconnect by monitoring the RSSI. How long is a disconnect event normally taking on your Nexus 6P? It's usually a couple seconds on my Android devices if it's an "unexpected" disconnect.

spiritonline, 2016-02-15:
He Rand,

Thanks for your quick reply! I've just timed it and a disconnect takes about 20 seconds while watching for an RSSI of 0 is almost instant. (~2 seconds max)
For my application it's critical that it responds swiftly to a disconnect. If i look at the android docs it says: "0 if no RSSI value is available." Could it be that this is only available for android 6 and thus the nexus 6p?

here's an image of console:
http://s14.postimg.org/vwtexsfgh/BLEstatus.png

randdusing, 2016-03-04:
Hey Jason,

I ran a few tests using a Polar H7 Monitor. I connected, began executing RSSI every 1 second and then physically detached the monitor. The monitor will stay on for several seconds and then the disconnected status will be received. Maybe not the best test since the device must be receiving a "disconnect event" from the Polar H7 before it shuts down.

Nexus 6 on 6.0.1 - I never see any 0 RSSIs and the RSSIs continue to vary up until the disconnect. 
Samsung Galaxy Tab A on 5.0.1 - Same as above.

I ran a test with another device that I can place into a faraday bag, which prevents the Bluetooth devices from sending a "disconnect event".

Nexus 6 on 6.0.1 - I never see any 0 RSSIs, but the RSSIs eventually stay the same for ~15 seconds until the disconnect is received.
Samsung Galaxy Tab A on 5.0.1 - Same as above.

I ran similar test to above, but I was continuously reading the device. It didn't affect how quickly the disconnect event fires.

In short, the 0 RSSI appears to be based on differences in hardware rather than Android versions. And it doesn't seem like the disconnect timeout can be changed on Android. You could possibly disconnect faster by detecting multiple RSSI calls of the same value over a period of X seconds, but that would probably lead to too many false positives.

closed 2016-03-09 by randdusing