randdusing/cordova-plugin-bluetoothle #237: Crash on IOS

kminkler, 2016-02-16:
Hi, have a weird issue with this plugin crashing but only immediately after the app starts.

After deviceready is fired in my cordova app, I'm calling in order...

isInitialized()
initialize()
isScanning()
startScan()
stopScan()
isConnected()
connect()
services()
characteristics() -- for each service
descriptors() -- for each characteristic
subscribe()
write()
...

I'm not sure it gets any farther than the isInitialized() or initialize() calls before it crashes the app and debugger (so it's hard to get details on where it's really dying), the UI hangs until the app just disappears... sometimes I've seen a few Memory warnings appear in lldb, as if it's stuck in some loop with a memory leak before it dies.

It seems like something in the native code isn't fully initialized by the time that I'm starting calls, because when I add a little timeout (e.g. 20 seconds), or initiate the sequence above from a button click, everything works perfectly fine, and it connects to the device without issues.

Because this happens right at app startup, and the debugger either can't quite connect in time or crashes along with the app, it's really hard to figure out where exactly this is crashing, but hopefully I can help diagnose this in some way.

randdusing, 2016-02-16:
Can you try not using isInitialized on the initial call? Also, can you post your deviceready / ble initialize code?

kminkler, 2016-02-16:
Tried that real quick, didn't seem to help.  I'll post the relevant code tonight, I need to spend some time extracting the relevant portions, there's a few separate files as it's an angular app with a service wrapper for a non-angular wrapper around the bluetoothle code.

kminkler, 2016-02-17:
This should be an excerpt of the relevant code... https://gist.github.com/kminkler/935f9246fcdad8788c45

randdusing, 2016-02-17:
Looks good to me... What's index.html look like, it's not getting automatically bootstrapped with ng-app? Are you using Ionic as well?

kminkler, 2016-02-17:
Right... no ng-app in the index.html, I updated the gist with an excerpt of the index.html.

I'm not using ionic, here's the versions of things I'm using..

cordova 6.0.0
cordova ios platform 4.0.1
xcode 7.2.1

cordova plugins:
com.jamiestarke.webviewdebug 1.0.8 "WebView Debug"
cordova-plugin-bluetoothle 3.0.1 "Bluetooth LE"
cordova-plugin-device 1.1.1 "Device"
cordova-plugin-geolocation 2.1.0 "Geolocation"
cordova-plugin-transport-security 0.1.1 "App Transport Security"
cordova-plugin-whitelist 1.2.1 "Whitelist"
phonegap-plugin-push 1.5.3 "PushPlugin"

randdusing, 2016-02-17:
And no crash on Android, right? It could possibly be related to another plugin. If you disable all but device, do you still get a crash?

kminkler, 2016-02-17:
Yeah, Android has always worked flawlessly, I've only had issue on IOS.

I removed all the plugins except for:
> $ cordova plugin list
>
> cordova-plugin-bluetoothle 3.0.1 "Bluetooth LE"
> cordova-plugin-device 1.1.1 "Device"

I was still able to replicate the issue.  With the timeout in place, the app loads, and connects to the device fine.

When the timeout is removed, and the scan is called right away, the app UI hangs, then the app ultimately crashes about a minute later.

kminkler, 2016-02-17:
I've also tried commenting out all the parameters to the scan call, in case something in there was causing the issue, but I get the crash even with no parameters:

>                    }, {
>                       /*  
>                        services: [ addresses.service ],
>                        'allowDuplicates': false,
>                        'scanMode': bluetoothle.SCAN_MODE_BALANCED,
>                        'matchNum': bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
>                        'matchMode': bluetoothle.MATCH_MODE_STICKY,
>                        'callbackType': bluetoothle.CALLBACK_TYPE_ALL_MATCHES,
>                        */
>                    });

randdusing, 2016-02-17:
Ahh, I just noticed you're using iOS 4.0.1. Do you get the error if you use 3.9.2 instead? I had some issues with 4.0.1. If that solves it, I'll try to take another look at 4.0.1 and see what the issue is.

kminkler, 2016-02-17:
so, downgrading to 3.9.2 does resolve the crash on startup.  However, the phonegap-plugin-push plugin requires at least 4.0.0... at version 3.9.2, that plugin crashes with a debugger error...

> Process 3849 stopped
> * thread #6: tid = 0x20bd31, 0x28285f16 JavaScriptCore`WTFCrash + 54, name = 'WebThread', stop reason = EXC_BAD_ACCESS (code=1, address=0xbbadbeef)     frame #0: 0x28285f16 JavaScriptCore`WTFCrash + 54
> JavaScriptCore`WTFCrash:
> ->  0x28285f16 <+54>: str    r1, [r0]
>     0x28285f18 <+56>: trap   
>     0x28285f1a <+58>: nop    
> 
> JavaScriptCore`WTFCrashWithSecurityImplication:
>    0x28285f1c <+0>:  push   {r4, r7, lr}

So I'm still at a bit of an impass... :)  I have a workaround with the 30-second pause that can get me by for the meantime though, until you find what's causing the issue in 4.x.

kminkler, 2016-02-17:
Downgrading to phonegap-plugin-push@1.5.1 solved the crash on cordova-ios@3.9.2

So I can stay on cordova-ios@3.9.2, which also seems to have solved another issue I was having with the app icon.

randdusing, 2016-02-17:
Yah, 4.0.1 has just been to buggy for me to warrant an upgrade so soon

randdusing, 2016-03-09:
I tried running the example app included as part of my Angular wrapper. It works well on Cordova iOS 4.1.0. You may want to try upgrading to that.

closed 2016-03-09 by randdusing