randdusing/cordova-plugin-bluetoothle #238: Scan issue

EkanshAg, 2016-02-17:
No device gets displayed, when scanning is done.

randdusing, 2016-02-17:
Do you have any devices advertising as Bluetooth LE devices?

slegouffe, 2016-02-18:
Hi ! Same for me, trying to scan for my Rasp 2 from my Nexus 5 on which the app is running. But nothing appear in console.log :-/
I'm unable to see the Rasp as everything seems to be ok. Note that I can pair my Nexus 5 and my Rasp when I use standard pairing (ie. directly from 6.0 android bluetooth menu). Here is my Rasp BlueTooth info : 
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 00:1A:7D:DA:71:13  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING PSCAN ISCAN 
	RX bytes:1516 acl:0 sco:0 events:77 errors:0
	TX bytes:1303 acl:0 sco:0 commands:77 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'xxxxxxxx'
	Class: 0x420100
	Service Classes: Networking, Telephony
	Device Class: Computer, Uncategorized
	HCI Version: 4.0 (0x6)  Revision: 0x22bb
	LMP Version: 4.0 (0x6)  Subversion: 0x22bb
	Manufacturer: Cambridge Silicon Radio (10) 

Thx

randdusing, 2016-02-18:
Android 6.0 requires location permissions to find unpaired devices. You will need to use hasPermission and requestPermission functions to ensure the permissions are good. Also, I'm not familiar with Raspberry Pi. Does it support Bluetooth, Bluetooth LE or both?

slegouffe, 2016-02-18:
Thx for your quick answer !  
Ok it works when I use "requestPermission", mean I can list my Mac...but still not get my RPi. 
I'm investigating about that and come back as soon as I found a solution.

slegouffe, 2016-02-19:
Hi ! 
Ok now I can see my Rasp ! Yeepppii !  
The tricks, on my pi,  I just add this command : 

`sudo hciconfig hci leadv 3`

So next step, pairing...

randdusing, 2016-03-09:
Anyone still having issues?

titoasty, 2016-03-20:
Yes I just tried today with API23, added requestPermission and everything, but no device appeared (although I have 3 ibeacons around me, perfectly detected on other apps)

randdusing, 2016-03-20:
You need to check hasPermission in addition to IsLocationEnabled

titoasty, 2016-03-20:
I already use requestPermission to ask for permission (which is done correctly).
I removed this line of code but after requestPermission the hasPermission returns true, so no problem on this side.

titoasty, 2016-03-20:
Found the problem !
So, it's not a permission problem, but location needs to be enabled (as you said). but I assumed it was activated with the request:true in my function call, which is a big misread.

Now, two questions :
  1) Why other BLE apps don't need to activate location to find ble devices ? (I tried with Locate)
  2) Is there a way to activate location with the plugin ? (perhaps another misread)

Thanks

titoasty, 2016-03-20:
Answering my own question, this should need a requestLocation function

randdusing, 2016-03-20:
There's no requestLocation function in the plugin. I looked into it previously but don't remember why I didn't include it. I can look into it again.

titoasty, 2016-03-21:
After some more tests, I can confirm that Location is definitely needed now with API 23, and so a requestLocation function may be very useful.

Location disabled : nothing returned
Location enabled : find all BLE devices

randdusing, 2016-03-23:
Plugin has a requestLocation function as of v3.3.0!

closed 2016-03-23 by randdusing

titoasty, 2016-03-23:
Thumbs up!
Thanks a lot