randdusing/cordova-plugin-bluetoothle #240: I need subscription explained

ricardoMogg, 2016-02-19:
Hi, I am clearly missing something about subscriptions. (I am sorry I am a noob at this). For what I understand once we subscribe to a device, it will notify our app about any changes in this characteristic right? What I am missing is how or where I will receive this change.

What I am currently doing is: 

$scope.read = function(characteristic) {
    var params = {address:$scope.address, service:$scope.service, characteristic:characteristic.uuid, timeout: 2000};

    $cordovaBluetoothLE.subscribe(params).then(function(obj) {
      console.log("Subscribe Auto Unsubscribe : " + JSON.stringify(obj));
    }, function(obj) {
      console.log("Subscribe Error : " + JSON.stringify(obj));
    }, function(obj) {
      console.log("Subscribe Success : " + JSON.stringify(obj));

      if (obj.status == "subscribedResult") {
        console.log("Subscribed Result");
      } else if (obj.status == "subscribed") {
        console.log("Subscribed");
      } else {
        console.log("Unexpected Subscribe Status");
      }
        
    });
  };

I am getting to Subscribed correctly, but after this I can't find anything to do with this subscription. Please help.

ricardoMogg, 2016-02-19:
Seems like I was having a bug on my device but it is now working.

randdusing, 2016-02-20:
Ahh, can you close the issue if it's good?

closed 2016-02-20 by ricardoMogg