randdusing/cordova-plugin-bluetoothle #244: Getting error while bluetoothle.startScan

MuraliKrishnaDev, 2016-02-29:
Hi,

I am getting error while bluetoothle.startScan,
Error is bluetoothle is not found,

Please guide me in this regard,

Thanks and regards,
Murali

randdusing, 2016-02-29:
Usually this means the plugin was installed wrong or it's being accessed before deviceready event is fired.

MuraliKrishnaDev, 2016-03-01:
Hi,

I tried after device ready event,I am using visul studio 2015 and cordova 5.4.0 is installed,

The following is the code,

Index.html
------------
<!DOCTYPE html>
<html>
<head>
    <meta charset="utf-8" />

    
    <meta http-equiv="Content-Security-Policy" content="default-src 'self' data: gap: https://ssl.gstatic.com 'unsafe-eval'; style-src 'self' 'unsafe-inline'; media-src *">
    <title>AndriodApps</title>

    
    <link href="css/index.css" rel="stylesheet" />
</head>
<body>
    <h1>Apps</h1>
   
    <div id="click" onclick="BluetoothConnection()">Turn Bluetooth</div>

    
    <script src="cordova.js"></script>
    <script src="scripts/platformOverrides.js"></script>

    <script src="scripts/index.js"></script>
</body>
</html>

index.js
---------
(function () {
    "use strict";

    document.addEventListener( 'deviceready', onDeviceReady.bind( this ), false );

    function onDeviceReady() {
        // Handle the Cordova pause and resume events
        document.addEventListener( 'pause', onPause.bind( this ), false );
        document.addEventListener( 'resume', onResume.bind( this ), false );
                document.getElementById('click').addEventListener('click', BluetoothConnection, false);
             // TODO: Cordova has been loaded. Perform any initialization that requires Cordova here.
    };
 function BluetoothConnection() {
        try {
        //cordova.plugins.BluetoothStatus.initPlugin();

            debugger
            //cordova.plugins.BluetoothStatus.enableBT();
            
            bluetoothle.initialize(function success() { }, function fail() { alert('error'); }, {
                "request": true,
                "statusReceiver": false
            });
     
            bluetoothle.requestPermission(function success() { }, function fail() { alert('error per'); });
            bluetoothle.startScan(function listHandler(scan) {
                alert(JSON.stringify(scan))
            }, function fail() { alert('error scanr'); }, {
                "services": [
                  "180D",
                  "180F"
                ],
                "allowDuplicates": true,
                 "scanMode": bluetoothle.SCAN_MODE_LOW_LATENCY,
                    "matchMode": bluetoothle.MATCH_MODE_AGGRESSIVE,
                    "matchNum": bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
                    "callbackType": bluetoothle.CALLBACK_TYPE_ALL_MATCHES,
            });
     
        } catch (e) {
            debugger;
            throw e;
        }
    }
)();

MuraliKrishnaDev, 2016-03-01:
Still I am getting same error,Please help me regarding this

Thanks and regards,
Murali

ZengYJ, 2016-03-01:
Hi,
I get the same error too.
I want to get the bluetooth list by scaning in app.
Bluetooth is initialized and enabled,then by startScan function no device could be found.
Actual I have turned on bluetooth of another phone(android 4.4).

Please help me or give some suggestion.
Thx.

ZengYJ, 2016-03-01:
Bluetooth version of two phones is 4.0+, support bluetoothle.

randdusing, 2016-03-01:
@MuraliKrishnaDev Try calling the scan in the initialize callback
@ZengYJ One of the devices must be advertising itself as a Bluetooth LE device. You can try using an app like LightBlue to emulate common BLE devices like heart rate monitors.

ZengYJ, 2016-03-01:
How to advertise itself as a buletooth le device?  Phone A which my app is installed on, acts  as server, and phone B which bluetooth is on. Could you give my some details. Thx

randdusing, 2016-03-01:
Currently the plugin only supports the central role. You'll need to use something like LightBlue to emulate BLE devices on the other phone.

ZengYJ, 2016-03-01:
I get it .
THX.

randdusing, 2016-03-09:
@ZengYJ If you have a spare iOS device, you can use 4.0.0-dev branch of the plugin to advertise the device as a peripheral.

randdusing, 2016-03-09:
@MuraliKrishnaDev Are you still having issues?

MuraliKrishnaDev, 2016-03-11:
Hi,
Still I am getting the issue,
Regards,Murali
Date: Wed, 9 Mar 2016 10:25:13 -0800

randdusing, 2016-03-11:
@MuraliKrishnaDev Can you try using the Ionic based example in the plugin's Angular wrapper? This should confirm if it's a coding issue or phone issue.

tfmontague, 2016-04-18:
@MuraliKrishnaDev 

Are you running the application from an Android phone?
You can't run this plugin on a windows desktop.

Please try following steps:
1. Download nodejs (https://nodejs.org/en/)
2. > npm install -g cordova
3. > cordova create "myapp"
4. > cordova platform add android
5. >  cordova plugin add cordova-plugin-bluetoothle
6. Write your app code
7. > cordova build android
8. Then upload your built app to the Android phone