randdusing/cordova-plugin-bluetoothle #247: Device not found

yog-he, 2016-03-09:
Hi,
I'm facing a new issue. 
var param = {"address":"7fa44329-90c9-3bc3-b34c-6c0f14ef5a06"};
When I call bluetoothle.connection(param); with iOS
I get {error: "connect", message: "Device not found", address: "7FA44329-90C9-3BC3-B34C-6C0F14EF5A06"}.

randdusing, 2016-03-09:
Do you see that device while scanning and then try to connect to it? Sounds like the device may just be turned off or out of range.

yog-he, 2016-03-09:
No, I scan first time to save the device address, and after I just use the saved address.

randdusing, 2016-03-09:
What type of device? Can you print out the scanResult object related to the address? I recommend checking out the example app in Angular Wrapper and trying to scan and connect.

yog-he, 2016-03-09:
What type of device?  :  iPhone 6 + 
Can you print out the scanResult object related to the address?
{
  "status": "scanResult",
  "rssi": -45,
  "name": "Test Device",
  "address": "7fa44329-90c9-3bc3-b34c-6c0f14ef5a06"
}
Could the address change between to iPhone?

randdusing, 2016-03-09:
Yah, the address will be different between devices you run your app on. You'll need to scan and connect from the same app if not doing so already.

yog-he, 2016-03-09:
Ok And In the same device do I need to scan everytime before do connection.

randdusing, 2016-03-10:
Nope, once you find the peripheral on the device, you shouldn't need to re-scan.

randdusing, 2016-03-23:
Still having this issue?