randdusing/cordova-plugin-bluetoothle #248: Android `bluetoothle.initialize` success callback not called after `bluetoothle.enable` invoked

niftylettuce, 2016-03-10:
Not sure what's going on here, but the Readme says that the `initialize` callback will get called again continuously whenever bluetooth is enabled/disabled - that doesn't seem to be the case though.  It only gets a success callback when it is first invoked, never again even if you do `bluetoothle.disable` and `bluetoothle.enable`.

randdusing, 2016-03-10:
What parameters are you passing to initialize()?

niftylettuce, 2016-03-10:
```js
cordova.window.bluetoothle.initialize(function(data) {
  // ...
}, function(err) {
  console.log('Initialize error', err);
}, {
  request: true,
  statusReceiver: true
});
```

niftylettuce, 2016-03-10:
the `function(data)` is only called once (I think you can infer that)

randdusing, 2016-03-10:
That looks correct. Just to confirm, what version of the plugin are you using?

niftylettuce, 2016-03-10:
Latest version

niftylettuce, 2016-03-10:
3.1.0

niftylettuce, 2016-03-10:
@randdusing btw I would gladly sponsor you to port this plugin to React Native, email me private if you want to discuss this, niftylettuce@gmail.com

niftylettuce, 2016-03-10:
Right now I'm using my fork of <https://github.com/axemclion/react-native-cordova-plugin> to use https://github.com/randdusing/cordova-plugin-bluetoothle so it's probably a bit messy, I really need a native React Native ("RN") bluetooth package for Android/iOS :laughing: .

randdusing, 2016-03-10:
Oh, the documentation is out of data with 3.0.0+, I just updated it. Initialize only needs the successCallback and params.

closed 2016-03-10 by niftylettuce