randdusing/cordova-plugin-bluetoothle #249: Slow read operation

t2wu, 2016-03-11:
While testing on HTC desire 816 ( Android 5) and HTC M9 ( android 6), I encountered a problem with read operation. After establishing connection, I did a read on one characteristic and write on another (a read-write pair). Then subsequent read-write pair was slow. Actually it's slow on the read, which will takes nearly two second after bluttoothGatt.readCharacteristics() before BlueToothGattCallbackExtends.onCharactersticRead gets called. 

The slowdown is not exhibited on the iPhone.

randdusing, 2016-03-12:
What type of peripheral device? I'll test this on my Android and iOS devices tomorrow.

randdusing, 2016-03-12:
I turned my iPad into a peripheral and then read the device name characteristic from the devices below. All of them took about the same time ~50-100ms. iPhone 5S on 9.2.1, Samsung Galaxy Tab A on 5.0.1 and Nexus 7 (2013) on 6.0.1.

Do you have any other brands? Perhaps it is an issue with the hardware HTC uses.

t2wu, 2016-03-15:
I tested it on Samsung S6 Edge, too. It is also slow. But I have not check thoroughly to see if it's out of a similar reason since it's someone else's phone. I guess I'll rest the issue for now. I will check later.

t2wu, 2016-03-29:
I was mistaken. We are using an HTC M8 instead of an M9. I stripped down our project so after device ready event and JQuery mobile ready event, we use the bluetoothle API directly, connect, request permission, discover service, then read every half a second indefinitely. The slow down very consistently starts at about 8 or 9 reads or so. The response to read delays by consistently 1 second. Since the onset of the delay and the duration of the delay is so consistent, it makes me wonder if it's throttled.

t2wu, 2016-03-29:
Interestingly, when we connect to another completely different device instead of the one in-house, the issue disappears! Perhaps it's chip-specific.

randdusing, 2016-03-29:
Do you wait for the readSuccess callback before starting another read?

t2wu, 2016-03-31:
Yes I did. It seems to do with handshake parameters being misread on the device's end. I don't think the problem is with the plugin. I guess you can close this issue.

closed 2016-03-31 by randdusing