randdusing/cordova-plugin-bluetoothle #251: Use ByteBuffer for JavaScript platforms instead of `window.atob` which isn't on all of them

niftylettuce, 2016-03-17:
Perhaps we should use https://github.com/dcodeIO/bytebuffer.js/wiki/API instead of relying on browser specific `btoa` and `atob` functions.

niftylettuce, 2016-03-17:
(I have this issue on React Native)

randdusing, 2016-03-17:
I haven't seen any browser specific issues with the btoa and atob functions. What exactly is the issue?

niftylettuce, 2016-05-17:
There's no issues - it's just that ByteBuffer is a much more mature library, and it also has a ton more helper functions.  Just a thought.

randdusing, 2016-07-09:
I'd rather not add external dependencies to the app. Ideally I would remove the whole encodedString process and just pass bytes as an array directly...

randdusing, 2016-07-09:
Enhancement: On next major version, redo how data is transferred. Base64 encoded strings add unnecessary confusion.