randdusing/cordova-plugin-bluetoothle #254: Android error after unexpected unload of page

phy25, 2016-03-24:
I have used `scanStop` on a unload event. Don't know whether that worked.

Reproduce: In a.html don't do anything, in b.html use `scan()`. During scan in b.html, press back button to go to a.html, and go into b.html again.

Error:

> I/chromium: [INFO:CONSOLE(312)] "Error in Error callbackId: BluetoothLePlugin1116574295 : TypeError: Cannot read property 'substr' of undefined", source: file:///android_asset/www/cordova.js (312)
> I/chromium: [INFO:CONSOLE(314)] "Uncaught TypeError: Cannot read property 'substr' of undefined", source: file:///android_asset/www/cordova.js (314)

Alse see this error repeatedly in some other situations:

> W/cr_AwContents: Application attempted to call on a destroyed WebView
                                                                     java.lang.Throwable
                                                                         at org.chromium.android_webview.AwContents.isDestroyed(AwContents.java:1148)
                                                                         at org.chromium.android_webview.AwContents.setNetworkAvailable(AwContents.java:2516)
                                                                         at com.android.webview.chromium.WebViewChromium.setNetworkAvailable(WebViewChromium.java:477)
                                                                         at android.webkit.WebView.setNetworkAvailable(WebView.java:802)
                                                                         at org.apache.cordova.engine.SystemWebViewEngine$1.setNetworkAvailable(SystemWebViewEngine.java:112)
                                                                         at org.apache.cordova.NativeToJsMessageQueue$OnlineEventsBridgeMode$2.run(NativeToJsMessageQueue.java:340)
                                                                         at android.app.Activity.runOnUiThread(Activity.java:5639)
                                                                         at org.apache.cordova.engine.SystemWebViewEngine$1.runOnUiThread(SystemWebViewEngine.java:116)
                                                                         at org.apache.cordova.NativeToJsMessageQueue$OnlineEventsBridgeMode.onNativeToJsMessageAvailable(NativeToJsMessageQueue.java:336)
                                                                         at org.apache.cordova.NativeToJsMessageQueue.enqueueMessage(NativeToJsMessageQueue.java:251)
                                                                         at org.apache.cordova.NativeToJsMessageQueue.addPluginResult(NativeToJsMessageQueue.java:240)
                                                                         at org.apache.cordova.CordovaWebViewImpl.sendPluginResult(CordovaWebViewImpl.java:307)
                                                                         at org.apache.cordova.CallbackContext.sendPluginResult(CallbackContext.java:63)
                                                                         at com.randdusing.bluetoothle.BluetoothLePlugin$31.onScanResult(BluetoothLePlugin.java:2299)
                                                                         at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:355)
                                                                         at android.os.Handler.handleCallback(Handler.java:743)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                         at android.os.Looper.loop(Looper.java:150)
                                                                         at android.app.ActivityThread.main(ActivityThread.java:5546)
                                                                         at java.lang.reflect.Method.invoke(Native Method)
                                                                         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:792)
                                                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:682)

randdusing, 2016-03-24:
How is your project set up? It looks like the plugin is trying to return a message to a WebView that has been destroyed.

phy25, 2016-03-31:
Sorry for the late reply. I cannot reproduce this anymore on the production code and the brand new debug-purpose project, so I will close this.

My project didn't modify the Android java code, so I also feel strange to see missing Webkit error.

closed 2016-03-31 by phy25

randdusing, 2016-03-31:
No worries, glad it's working out now.