randdusing/cordova-plugin-bluetoothle #255: Syntax error in plugin.xml

chriswoodle, 2016-03-24:
Seems to be an error on line 66

randdusing, 2016-03-24:
What version of Cordova are you on? And what platforms do you have added?

chriswoodle, 2016-03-24:
I'm using visual studio tools for apache cordova, Cordova version 6.0.0, android, ios, windows

[image]

I removed the "<" and it worked fine.

randdusing, 2016-03-24:
Hmm, it's not valid XML according to an online XML validator, but Cordova's documentation says it should work: https://cordova.apache.org/docs/en/5.0.0/plugin_ref/spec.html See the config-file section.

Is it just a validation error within VS, or does it prevent the app from building?

chriswoodle, 2016-03-24:
Agreed,
Using visual studio to build it does fail building, (Character '<', hexadecimal value 0x3c is illegal in XML attribute values.)

Seems to build fine from the cli (6.1.0)

randdusing, 2016-03-24:
May want to submit a bug to VS Tools for Cordova

chriswoodle, 2016-03-24:
Ok, sounds good

closed 2016-03-24 by chriswoodle