randdusing/cordova-plugin-bluetoothle #256: Dev branch server side support

roeku, 2016-03-25:
Hi, could you perhaps explain how this would work. I've installed the angular example app of the dev branch but I'm not exactly sure how to proceed in connecting multiple (android) devices to one iBeacon. 

I tried to:
- initialize
- start scan
- stop scan
- connect to the iBeacon (with the first device, which works)
- connect to the iBeacon (with the second device, which fails (timeout))

Connecting using the devices separately works though.

Also when I
 `bower install git://github.com/randdusing/ng-cordova-bluetoothle.git#v4.0.0-beta.1`
I didn't see an example folder in my directory.

randdusing, 2016-03-25:
Are you creating an iBeacon with the server side code? If it's a normal iBeacon, it may not be able to connect to multiple devices.

I can't recreate the issue with bower and the example folder not coming through. Just cleared out my bower library and ran bower install again. Everything came across.

roeku, 2016-03-25:
I'm just going to explain what I'm doing to clear things up:

I have a parrot miniDrone of which I'm trying to read out the RSSI using multiple devices(android, iphone). 
When the drone is near one of these devices it receives a signal to stop moving or start. When I do this using the startScan function to receive the RSSI (without connecting) the refresh rate is too slow. So I'm trying to connect to the drone with multiple devices to increase the rate I receive the RSSI. I have also equipped myself with estimote beacons but their ranging is also limited to max 1 second refresh rate.

randdusing, 2016-03-25:
How are you making your scan call? Could be some improvements on that side specifically allowDuplicates => true and then the Android 21/23 options for more "aggressive" scanning.

4.0.0's main improvement is allowing you to turn your device into a Bluetooth LE peripheral (like a heart rate monitor). I don't think it will help in situations like yours.