randdusing/cordova-plugin-bluetoothle #263: Queue Subscribes

RangerMauve, 2016-04-04:
I've been using the master branch for connecting to my device, and reading/writing seems to queue up just fine. However, I'm experiencing issues subscribing to a certain notification. Specifically, it errors out [here](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/android/BluetoothLePlugin.java#L1589) saying that it was "Unable to write descriptor". Now, I have all my writes/reads/subscribes happening at almost the same time because this was handled properly [with queuing](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/BLECentralPlugin.java#L271) in the library I had previously worked with.

I'm assuming it's an issue with the `subscribe` failing because either one of my reads or writes are happening at the same time it's attempting to write to the descriptor.

I'm testing this on Android 5.1.1, with cordova platform 5.1.1.

Would a PR be welcome to add this feature? Are there any workarounds I can take in the meantime? (I'm thinking I'll just add an artificial delay to the subscribe call in the API wrapper I made)

randdusing, 2016-04-05:
@RangerMauve Thanks for trying out the master branch. A PR would be welcome, but I may be able to work on it over the next few days. It should be pretty straightforward to recreate the functionality for read/writeDescriptors and subscribe/unsubscribe.

In the meantime, a delay is your best option even if it's a bit hacky. If you're using the Angular wrapper, you could also try chaining the promises together.

RangerMauve, 2016-04-06:
Adding a delay of a few seconds seems to have done it for now. I'll see if I have time to allocate to this tomorrow if you haven't gotten to it by then. Lots of other stuff piling up at the same time, sadly. :P

randdusing, 2016-04-06:
I'm making some updates as we speak and will be testing later today. If all goes well, should see it merged to master branch sometime tonight.

RangerMauve, 2016-04-06:
Perfect. Then I'll give it a field test tomorrow morning.

randdusing, 2016-04-06:
@RangerMauve I updated master with queueing support for all the operations. Let me know if you run into any issues.

closed 2016-04-06 by randdusing

RangerMauve, 2016-04-07:
Bleh, I have to take care of some other things first. Will probably have some time later this afternoon to devote to this again.

RangerMauve, 2016-04-13:
I've got the code currently in master running out in the wild for a couple of days now and it's all running smoothly as far as I can tell. Thanks!

randdusing, 2016-04-13:
Cool, glad it's working out!