randdusing/cordova-plugin-bluetoothle #266: subscribed failed comming back with "isDisconnected" error out of no where?

kgillespieanaren, 2016-04-07:
So I am suddenly getting a new error with my phonegap app using the BLE plugin.

I subscribe to one characteristic in my app and my subscribed successful callback goes of as expected, with status of "subscribed".

I never call another subscribe.

After waiting about 30 seconds the subscribed failed callback will get an "isDisconnected" error completely unprovoked.

Going to see if not subscribing prevents this but this is a very big problem.

randdusing, 2016-04-07:
Is the device disconnecting? If the device disconnects, any pending
callbacks will return isDisconnected.

kgillespieanaren, 2016-04-07:
The device claims that the call to disconnect is coming from the central side.

randdusing, 2016-04-07:
What type of devices on the central side? And is the subscribe actually
sending data? I noticed while testing my client and peripheral code between
two iOS devices that the client would disconnect. This would happen after
operations stopped executing and the connection became idle. I wonder if
the central is disconnecting at a lower level. Either way the new code
shouldn't have caused the disconnect, just reporting back the
isDisconnected state of the subscription.

kgillespieanaren, 2016-04-07:
Well it's an Android device for central using the plugin and a BCM20737 based embedded chip set.

We haven't changed anything on the embedded side and we have updated the Phonegap BLE plugin to the latest version from a build back from one at the end of February. It's seem that I can keep my connection stable by not subscribing. I will try and gather more data, but this may also be happening with iOS devices as well.

randdusing, 2016-04-07:
What version were you on before? Any differences in Android version? I can
create a dev branch with the changes removed and see if it affects it.

kgillespieanaren, 2016-04-07:
Never mind, it simply took longer to disconnect even after disabling my subscribe. I'll gather some more information before bothering you further. It seems though that the connection has become very unstable on both Android and iOS in the last two months. Will try and pin point which commit for you.

randdusing, 2016-04-08:
No worries. The core connection functionality hasn't really changed, so I'm
not sure what's going on. It could also be related to newer Android or iOS
versions

RangerMauve, 2016-07-19:
Lately we've been encountering an issue where after a while our peripherals randomly start reporting that they've been disconnected right after they've connected when we try to configure them with writes and reads. We've checked to see if maybe our JS application logic was somehow screwing up, but when we re-launch the JS logic from a fresh state, it seems that the issues still persist. As well, when we relaunch the app itself, things seem to go smoothly again. Restarting the peripherals also had no effect. So far we're not sure how to reproduce it other than "Wait a while and maybe it'll happen". Seems a little similar to the issue described here. Only really started seeing it after our clients upgraded to Android 6, but BLE was totally unstable before that so I can't say for sure it wasn't happening before.

acutetech, 2017-07-21:
You say: " If the device disconnects, any pending callbacks will return isDisconnected." I found this, when either the central or the peripheral disconnects. Could I suggest it is worth adding this to the .subscribe() documentation, as a feature?

randdusing, 2017-07-30:
Will update shortly.