randdusing/cordova-plugin-bluetoothle #268: Xcode Thread warning - Plugin should use background thread

tfmontague, 2016-04-13:
Does anyone know why Xcode produces the following warning in the console log? Seems to occur while the iOS plugin is starting a bluetooth scan.

THREAD WARNING: ['BluetoothLePlugin'] took '33.505859' ms. Plugin should use a background thread.

tfmontague, 2016-04-13:
Came across this StackOverflow post -
http://stackoverflow.com/questions/24572174/phonegap-media-plugin-should-use-a-background-thread

Anyone else experiencing this warning? Think the solution might be to wrap the blocking `sendPluginResult` command, in a `runInBackground` callback as mentioned in the previous StackOverflow. Can someone confirm if this is correct?

```
[self.commandDelegate runInBackground:^{
  //Send scan started status
  NSDictionary* returnObj = [NSDictionary dictionaryWithObjectsAndKeys: statusScanStarted, keyStatus, nil];
  CDVPluginResult* pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_OK messageAsDictionary:returnObj];
  [pluginResult setKeepCallbackAsBool:true];
  [self.commandDelegate sendPluginResult:pluginResult callbackId:scanCallback];
}];
```

randdusing, 2016-04-13:
Do you get the same error on any newer devices? I'm not getting it on an iPhone 5S with my example app. Is the app doing anything else that could be putting load on the device?

tfmontague, 2016-04-19:
I get the warning with the [ng-cordova-bluetoothle](https://github.com/randdusing/ng-cordova-bluetoothle) app as well. It happens for an app that has been built and installed. On a rebuild (without installation) the warning goes away. It might be an iPhone 4S issue, but the warning makes sense to me. I don't think bluetooth communication should be on the main thread.

> Plugin methods ordinarily execute in the same thread as the main interface. If your plugin requires a great deal of processing or requires a blocking call, you should use a background thread.
http://docs.phonegap.com/en/edge/guide_platforms_ios_plugin.md.html#iOS%20Plugins_threading

randdusing, 2016-04-19:
I don't think there's really much processing going on under normal usage, and none of the calls should be blocking too long. Adding multiple threads makes things much complicated since I'll need to deal with race conditions between threads.

Do you see the warning on any calls outside of startScan?

tfmontague, 2016-04-26:
I got the thread warning again, in Xcode console. This time on bluetooth initiation (client).

> THREAD WARNING: ['BluetoothLePlugin'] took '149.864990' ms. Plugin should use a background thread.

Steps to reproduce warning:
1. Turn off bluetooth on your iPhone
2. Install the app
3. An alert will state "Bluetooth is off" and ask you to enable it. Take no action, just leave the alert on the screen for about 2 minutes.
4. Afterwards click "Settings" button, and enable bluetooth

tfmontague, 2016-04-26:
Also got the warning (a couple times) from `bluetoothle.connect()`, when the status is `disconnected`.

randdusing, 2016-05-02:
Hmm, not able to recreate the warning via the settings prompt either, but my devices are a little newer.

closed 2016-06-20 by tfmontague