randdusing/cordova-plugin-bluetoothle #269: iOS - Memory Warning

tfmontague, 2016-04-13:
Cordova app doesn't load, and results in a memory warning when ` bluetoothle.startAdvertising()` and `bluetoothle.startScan()` when used in the same app. Testing on iPhone 4S, running iOS 9.1.

Xcode console prints:
> Received memory warning.

randdusing, 2016-04-13:
Perhaps related to #268 ? Is the app doing anything else that may be putting extra load on the device?

tfmontague, 2016-04-19:
I think the problem is that the Central and the Peripheral can't be initialized at the same time. And I think that they fight for the same resource, which results in a Memory warning. I was able to to fix the issue, by invoking the `initialize` function after the `initializePeripheral` callback returned.

The [ng-cordova-bluetooh](https://github.com/randdusing/ng-cordova-bluetoothle) app is built in such a way that the user needs to toggle between Central and Peripheral views, and therefore it's unable to test this edge case - i.e. when `initialize` and `initializePeripheral` buttons are pressed together.

**This works!**
```
var cbSuccess = function () {
  bluetoothle.initialize(...);
};
...
bluetoothle.initializePeripheral(cbSuccess, cbError, params); 
```

**This doesn't work!**
```
bluetoothle.initializePeripheral(...); 
bluetoothle.initialize(...);
```

randdusing, 2016-04-19:
I recommend the first way. In one of the examples of the example app, I call initialize and then initializePeripheral in the success, so I never ran into the memory warning issue. Probably the best thing I can do now is add this to the documentation.

randdusing, 2016-05-02:
Hmm, I'm not able to recreate this on my iPad Air or iPhone 5S. Can you see how much memory is being used? The memory doesn't change between my default example app, and the example app where initialize and initializePeripheral are automatically called back to back.

tfmontague, 2016-05-06:
It's clear there's a memory leak.

**This works!**
```
var cbSuccess = function () {
  bluetoothle.initialize(...);
};
bluetoothle.initializePeripheral(cbSuccess, cbError, params); 
```
[image]

**This doesn't work!**
```
bluetoothle.initializePeripheral(...); 
bluetoothle.initialize(...);
```
[image]

tfmontague, 2016-05-06:
I wonder if it's my version of Xcode (I'm using a beta 3 version of 7.1)... the odd part is that when I use the Instrument app to profile the Leaks (rather than the memory chart provided by Xcode) - there aren't any leaks found... and the app seems to magically work afterwards.
[image]

randdusing, 2016-05-07:
I'm using Version 7.3 (7D175). Curious if you encounter the same issue if you update.

closed 2016-06-20 by tfmontague