randdusing/cordova-plugin-bluetoothle #270: Airplane mode doesn't signal disconnects (Android)

RangerMauve, 2016-04-15:
When Airplane Mode (a.k.a flight mode) is enabled on Android, even though all Bluetooth devices get disabled, the disconnection callbacks aren't invoked by the plugin.

The main problem is that the application ends up thinking that the devices are connected even though they aren't.

Any chance the plugin could handle calling the disconnect callbacks? Or would you have any advice on how to handle this sort of scenario?

Thanks in advance! :D

randdusing, 2016-04-15:
You could handle it via the initializeSuccess callback returning disabled currently, but it's not the best approach. Is this only on Android? I haven't tested this behavior in a while, so I don't remember how it works.

Ideally Bluetooth being disabled should trigger the disconnected event for all current connections and any pending operations.

RangerMauve, 2016-04-15:
Well, I've only tested it on android at the moment. It's really when a user disables it at runtime that we see the issue. When Bluetooth gets disabled though the plug-in it seems to properly register that the devices are disconnected, but I'll need to do some more testing on Monday. I'll also try the stuff with initializeSuccess

randdusing, 2016-04-20:
Happens on iOS as well. Hopefully I can get to this week.

randdusing, 2016-05-02:
I'm working on the iOS implementation right now. Android should be a pretty similar fix. There's one issue I'm running into when it comes to managing the connection callback, but I should have it figured out shortly.