randdusing/cordova-plugin-bluetoothle #271: BluetoothLePlugin.m at line of 817 method Deprecated

coolnameismy, 2016-04-18:
BluetoothLePlugin.m at line of 817
````
  NSArray* peripherals = [centralManager retrieveConnectedPeripheralsWithServices:serviceUuids];
````

retrieveConnectedPeripheralsWithServices:servics did Deprecated in iOS 7.0 ,it may unsoppout for iOS 9.0+

randdusing, 2016-04-18:
Where are you seeing that this function is deprecated? I don't see it in the [documentation](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/#//apple_ref/occ/instm/CBCentralManager/retrieveConnectedPeripheralsWithServices:) or when I compile an app.

coolnameismy, 2016-04-19:
i find it im code

发自我的 iPhone

randdusing, 2016-04-19:
What iOS version are you targeting?