randdusing/cordova-plugin-bluetoothle #272: Queue connects

RangerMauve, 2016-04-22:
Currently I seem to be running into issues when I attempt to connect to multiple devices at once. It'd be nice if this operation could also be queued.

randdusing, 2016-04-22:
What type of device?

RangerMauve, 2016-04-22:
We're using a BLE button called [v-alert](http://www.vsnmobil.com/products/v-alrt), basically we have a dozen machines attempting to connect devices within range while coordinating with each other. The issue we're seeing is that a given machine may be told to connect to like four devices at once depending on which one scans which devices first.

randdusing, 2016-04-22:
Sorry, I mean what type of device are you running into the issue on? Android vs iOS, platform version, etc?

RangerMauve, 2016-04-22:
Whoops, it's android on a Samsung Tab E. Running 5.something

randdusing, 2016-04-24:
And what exactly is the error? Does it return an error immediately?

RangerMauve, 2016-04-24:
No, what seems to happen is the Bluetooth stack just craps out. We can no longer scan devices even though scanning doesn't cause any errors, devices become disconnected without the callbacks getting invoked. Stuff like that. Not 100 percent sure that this is causing it though, more just a hypothesis we had.

randdusing, 2016-04-24:
Some Android devices seem to have especially flakey Bluetooth stacks. I may be able to test an Android devices with 5+ devices. But I'm not sure if I'll be able to do much if there's an issue with the BT stack at a lower level. Do you have access to any newer Android devices or ones on 6.0? I'm curious if you have the same issues.

randdusing, 2016-04-24:
Are you using the latest master commit that queues read/write operations?

RangerMauve, 2016-04-24:
Well, on older devices with 4.3 we actually haven't seen this issue manifest as much, it would still happen sometimes, but either after many days of activity or if we had Bluetooth being turned on and off frequently. Sadly, these devices are deployed to a facility and we can't exactly replace them at the moment. We recently acquired a 6.0 phone, but haven't tested its Bluetooth yet.

Specifically we're running this on the Samsung Tab A.

I think [this issue in Android](https://code.google.com/p/android/issues/detail?can=2&start=0&num=100&q=&colspec=ID%20Status%20Priority%20Owner%20Summary%20Stars%20Reporter%20Opened&groupby=&sort=&id=191831) is related to the problem.

And yes, the read/write queuing is working perfectly. :D

randdusing, 2016-05-02:
I have a Samsung Tab A as well but haven't really tested with more than two device connections. I have a bunch of devices, so I may be able to recreate the issue.

I know I can get the Bluetooth stack into a really unreliable state if I turned the Tab A into an advertising device.

RangerMauve, 2016-05-02:
We have a setup with about 15 tablets and around 12 BLE devices that they're connecting to, with BLE devices disconnecting and switching owners frequently. It's a bit hard to reproduce at small scales, sadly.

Currently we have people specialized in the tablets looking into it, so hopefully we'll see some results. I'll post here if we have any clear path to avoiding the issue or if it gets patched by Samsung.

randdusing, 2016-05-02:
Cool, thanks for the update. Hopefully a workaround is discovered.

RangerMauve, 2016-05-05:
I think the issue we're seeing may be caused by #280 Not 100% sure yet.

closed 2016-05-05 by RangerMauve

randdusing, 2016-05-07:
Oh, were you actually encountering a crash?

RangerMauve, 2016-05-07:
Well, I'm still not convinced that this is the only issue we had, but it's definitely one thing that was found for sure.