randdusing/cordova-plugin-bluetoothle #273: iOS - Unrecognized Selector Error

tfmontague, 2016-04-26:
Got the following error from either `bluetoothle.services()` or `blutoothle.characteristics()`. Still pin-pointing what might have caused the error.

> -[NSNull compare:options:range:]: unrecognized selector sent to instance 0x36795e48
> 2016-04-25 23:14:19.209 myApp[202:7262] *** WebKit discarded an uncaught exception in the webView:decidePolicyForNavigationAction:request:frame:decisionListener: delegate: <NSInvalidArgumentException> -[NSNull compare:options:range:]: unrecognized selector sent to instance 0x36795e48

Environment: Xcode (v7.1), iPhone 4S, iOS 9.1

tfmontague, 2016-04-26:
Looks like `bluetoothle.characteristics` is throwing the error.

tfmontague, 2016-04-27:
Found the issue. Passed an undefined UUID for the characteristic.

var params = {
  address: "5163F1E0-5341-AF9B-9F67-613E15EC83F7",
  service: "1234",
  characteristics: [undefined]
};
bluetoothle.characteristics(..., params);

tfmontague, 2016-04-27:
For app stability could implement a schema on the parameters being passed through bluetoothle.js.
https://github.com/molnarg/js-schema

randdusing, 2016-04-27:
I'll see if the error can be caught within the plugin itself. It shouldn't be left uncaught.

randdusing, 2016-05-02:
Latest commit in master branch will now check for invalid UUIDs better

closed 2016-05-02 by randdusing

tfmontague, 2016-05-07:
There's still a lot of errors I can throw in your function calls
- Not including parameters
- Including parameters but with the wrong key names
- Including parameters with with the wrong value types (string, number, undefined, null)

It's much more effective to implement a schema at the javascript layer (bluetoothle.js) to catch malformed parameters. Handling malformed parameters at the iOS native layer is not a scalable solution. For every new platform you add, you'll need to write code to handle the parameters. That violates DRY (Don't Repeat Yourself) methodology, and it increases the size of the plugin.

randdusing, 2016-05-08:
PRs are welcome. Right now I'm more focused on bug fixes and improved BLE functionality coverage since my time is limited.