randdusing/cordova-plugin-bluetoothle #275: iOS - Read Errors

tfmontague, 2016-04-30:
Getting various read errors on iPhone 4S (iOS 9), when trying to read `value`.

```
{"service":"1234","name":"iPhone","characteristic":"ABCD","error":"read","address":"7704EB86-C3AB-65AA-5753-79307197AE94","message":"Error Domain=CBATTErrorDomain Code=241 \"Unknown ATT error.\" UserInfo={NSLocalizedDescription=Unknown ATT error.}"}

{"service":"1234","name":"iPhone","characteristic":"ABCD","error":"read","address":"A80E4B89-4307-E80B-8449-6492DC4D0166","message":"Error Domain=CBATTErrorDomain Code=10 \"The attribute could not be found.\" UserInfo=0x16554620 {NSLocalizedDescription=The attribute could not be found.}"}

{"service":"1234","name":"iPhone","characteristic":"ABCD","error":"read","address":"79F5B1B7-739D-C27A-43DE-FAE3C5E62FCA","message":"Error Domain=CBErrorDomain Code=0 \"Unknown error.\" UserInfo=0x16d72230 {NSLocalizedDescription=Unknown error.}"}
```

```
var params = {
  service: "1234",
  characteristics: [{
    uuid: "ABCD",
    permissions: {
      read: true
    },
    properties: {
      read: true
    },
    value: bluetoothle.bytesToEncodedString(
      bluetoothle.stringToBytes("characterValue"))
  }]
};

bluetoothle.addService(..., ..., params);
```

randdusing, 2016-04-30:
I love iOS's descriptive error messages. I know I've seen this error before, but never figured out a specific cause.

tfmontague, 2016-04-30:
**Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found."**
Looks like it's related to notifications and `setNotify`.

- https://forums.developer.apple.com/thread/20115
- http://stackoverflow.com/questions/31275013/subscribing-for-notifications-from-a-cbcharacteristic-does-not-work/31563121
- https://www.b4x.com/android/forum/threads/ble-objective-c-sdk-for-dfrobot-bluno.57750/

randdusing, 2016-04-30:
Can you post the code for the client side?

tfmontague, 2016-04-30:
It's fairly bland - but here's the code.

1. Init client
2. Scan
2. Connect
3. Discover the services
4. Discover the characteristics
5. Read

```
var addressUuid = null;
var serviceUuid = "1234";
var characteristicUuid = "ABCD";

function init() {
  bluetoothle.initialize(
    function (response) {if (response.status === "enabled") {scan();}},
    {request: true, statusReceiver: true, restoreKey: "bluetoothleplugin"}
  );
}

function scan() {
  bluetoothle.startScan(
    function (response) {if (response.status === "scanResult") {address = response.address; connect();}},
    function () {},
    {services: [serviceUuid], allowDuplicates: true}
  );
};

function connect() {
  bluetoothle.connect(
    function (response) {if (response.status === "connected") {services();}},
    function () {},
    {address: addressUuid}
  );
};

function services() {
  bluetoothle.services(
    function () {characteristics();},
    function () {},
    {address: addressUuid, services: [serviceUuid]}
  );
};

function characteristics() {
  bluetoothle.characteristics(
    function () {read();},
    function () {},
    {address: addressUuid, service: serviceUuid, characteristics: [characteristicUuid]}
  );
};

function read() {
  bluetoothle.read(
    function (response) { console.log(response);},
    function () {},
    {address: addressUuid, service: serviceUuid, characteristic: characteristicUuid}
  );
};

```

tfmontague, 2016-05-01:
Also here's the server code

1. init server
2. add service
3. advertise service
4. (optional) respond to read

```
var addressUuid = null;
var serviceUuid = "1234";
var characteristicUuid = "ABCD";

initPeripheral();

function initPeripheral() {
  bluetoothle.initializePeripheral(
    function (response) {
      if (response.status === "enabled") {addService();}
      if (response.status === "readRequested") {respond();}
    },
    function () {},
    {request: true, restoreKey: "bluetoothleplugin"}
  );
}

function addService() {
  bluetoothle.addService(
    function () {if (response.status === "serviceAdded") {advertiseService();},
    function () {},
    {service: serviceUuid,
       characteristics: [{
          uuid: characteristicUuid,
          permissions: {read: true},
          properties : {read: true},
          value: bluetoothle.bytesToEncodedString(bluetoothle.stringToBytes("characterValue"))
        }]}
  );
}

function advertiseService() {
  bluetoothle.startAdvertising(
    function () {},
    function () {},
    {services: [serviceUuid], name: "myApp"}
  );
}

function respond() {
  bluetoothle.respond(
    function (response) {console.log(response);}
    function () {}
    {requestId: 0,
      value: bluetoothle.bytesToEncodedString(
        bluetoothle.stringToBytes("respondValue"))
    }
  );
}
```

tfmontague, 2016-05-01:
This may share some relation to #278 . Adding `notify: true` may resolve the issue (haven't tested this enough to confirm).

```
var params = {
  service: "1234",
  characteristics: [{
    uuid: "ABCD",
    permissions: {
      read: true
    },
    properties: {
      read: true,
      notify: true
    },
    value: bluetoothle.bytesToEncodedString(
      bluetoothle.stringToBytes("characterValue"))
  }]
};

bluetoothle.addService(..., ..., params);
```

randdusing, 2016-05-01:
That's weird that ```notify: true``` would be required. I'll see if I can recreate with the example app.

randdusing, 2016-05-02:
This may be an issue with the requestId being hardcoded to zero rather than the value received in the read/writeReceived event. I set the service to only have read properties and permissions, and the read worked fine.

closed 2016-06-20 by tfmontague