randdusing/cordova-plugin-bluetoothle #277: iOS - Read Error: Code 241

tfmontague, 2016-04-30:
`{"service":"1234","name":"iPhone","characteristic":"ABCD","error":"read","address":"7704EB86-C3AB-65AA-5753-79307197AE94","message":"Error Domain=CBATTErrorDomain Code=241 \"Unknown ATT error.\" UserInfo={NSLocalizedDescription=Unknown ATT error.}"}`

tfmontague, 2016-05-01:
This might be caused by the Peripheral not responding to a read request in time...

http://stackoverflow.com/questions/33299794/why-does-it-take-longer-for-cbcharacteristics-to-receive-a-write-value-than-it-d

randdusing, 2016-05-01:
Are you connecting to a device that's using the server side of the plugin? Are you sure it's responding to the proper requestId?

tfmontague, 2016-05-02:
I've only been able to generate this error once. So it's not as common as the #275 and #278 errors.
For all three errors seen in #275, #277, and #278. The setup is the same.

Device A - runs a client
Device B - runs a server

The client and server connect, services and characteristics on the server are discovered, and at times the client will `read` and the server will `respond`. Most the time

Haven't researched the `requestId` to understand the role it plays in `read`. But I'm using `requestId: 0`, as seen in the server code that I provided in #275 .

randdusing, 2016-05-02:
Ahh, you need to use the requestId that is provided in readRequested or writeRequested event.

tfmontague, 2016-05-02:
Got it! I'll try that out.

closed 2016-06-20 by tfmontague