randdusing/cordova-plugin-bluetoothle #278: iOS - Read Error: Code 0

tfmontague, 2016-04-30:
`{"service":"1234","name":"iPhone","characteristic":"ABCD","error":"read","address":"79F5B1B7-739D-C27A-43DE-FAE3C5E62FCA","message":"Error Domain=CBErrorDomain Code=0 \"Unknown error.\" UserInfo=0x16d72230 {NSLocalizedDescription=Unknown error.}"}`

tfmontague, 2016-05-01:
This error also looks like it's related to notifications and `setNotify`.

**Error Domain=CBErrorDomain Code=0**
http://stackoverflow.com/questions/9545500/notifications-on-corebluetooth-returning-a-cberrordomain-code-0

> I think some of these errors occur when you try to read a value that is marked to only be available through notifications rather than read directly.

tfmontague, 2016-05-01:
Setting `notify: true` in properties seems to fix this error.

```
var params = {
  service: "1234",
  characteristics: [{
    uuid: "ABCD",
    permissions: {
      read: true
    },
    properties: {
      read: true,
      notify: true
    },
    value: bluetoothle.bytesToEncodedString(
      bluetoothle.stringToBytes("characterValue"))
  }]
};

bluetoothle.addService(..., ..., params);
```

randdusing, 2016-05-01:
How often are you changing the characteristics on the server side? I wonder if the client is caching the characteristics' configurations and causing issues.

tfmontague, 2016-05-02:
If you mean "changing" as editing the code, the answer is frequently. But adding `notify: true` fixed the issue right away. Didn't seem like it was cached.

randdusing, 2016-05-02:
Were you trying to read or subscribe from the client side?

tfmontague, 2016-05-02:
I have a client app with the code needed to read or subscribe. If I want to read I comment out the subscribe call. If I want to subscribe I comment out the read call.

I got the Code 0 error for both `read` and `subscribe` on different builds of the client app. It went away for `subscribe` after adding `notify: true`; for `read` it never went away (maybe due to the hard-coded requestId).

All three of these read errors ( #278, #241, #275 ) may have been caused by hard-coding the `requestId` for `respond`. I've fixed the RequestId so it's no longer hard-coded, but I can't verify if that fixed the read errors. Haven't spent enough time with the app after making this code change to conclude. But I will try to give an update in a day or two.

closed 2016-06-20 by tfmontague