randdusing/cordova-plugin-bluetoothle #280: Race condition could cause

RangerMauve, 2016-05-05:
We've been having issues with bluetooth crapping out after a while and sometimes with our application crashing unexpectedly. After some monitoring we saw the following stack trace within logcat:

```
java.lang.NullPointerException: Attempt to invoke virtual method 'void org.apache.cordova.CallbackContext.sendPluginResult(org.apache.cordova.PluginResult)' on a null object reference
               at com.randdusing.bluetoothle.BluetoothLePlugin$3.onScanResult(BluetoothLePlugin.java:2583)
               at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:393)
               at android.os.Handler.handleCallback(Handler.java:739)
               at android.os.Handler.dispatchMessage(Handler.java:95)
               at android.os.Looper.loop(Looper.java:145)
               at android.app.ActivityThread.main(ActivityThread.java:6117)
               at java.lang.reflect.Method.invoke(Native Method)
               at java.lang.reflect.Method.invoke(Method.java:372)
               at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
               at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)
```

The issue is that `scanCallbackContext` is null at the time the plugin is receiving a new scanned notification from Android.

After inspecting [the code](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/android/BluetoothLePlugin.java#L2570) it would appear to be impossible to have this happen since there is a null check at the beginning of the callback.

However, I think this may be related to a hard-to-reproduce race condition due to the fact that we star/stop scanning on a different thread than what is used for receiving scanned devices.

I'm thinking the flow looks something like this:

| plugin thread | bluetooth thread |
| --- | --- |
| startScan | beginScanning |
| -- | scanned device |
| -- | check null |
| stopScan | set up result |
| -- | sendPluginResult|
| -- | explode |

I propose that starting / stopping scans, as well as reacting to scanned devices make use of the `synchronized` keyword to ensure only one thread can mess with that at a time.

Would a PR be welcome for this?

randdusing, 2016-05-07:
I thought the latest plugin version where separate threads were removed would fix this... I'll try to check out the PR as soon as possible.

RangerMauve, 2016-05-07:
I'm guessing that bluetooth is running on a different thread than the plugin. Which is why it should be synchronized.

PR #281 (2016-05-11): Syncronize scanning-related android code
Fixes #280
closed 2016-05-11 by randdusing