randdusing/cordova-plugin-bluetoothle #282: Read is sent 4 responses!

tfmontague, 2016-05-07:
Environment - iPhone 4S running iOS 9

Have a client that reads.
```
// Client
bluetoothle.read();
```
And a server that responds.
```
// Server
var parameters = {
  requestId: <integer>
  value: 'Lorem ipsum dolor sit amet, consectetur adipiscing elit. Integer nec odio. Praesent libero. Sed cursus ante dapibus diam. Sed nisi. Nulla quis sem at nibh elementum imperdiet. Duis sagittis ipsum.'
};
bluetoothle.respond(..., parameters);
```

When I set a value that is 208 bytes.
It gets truncated to 157 bytes, and it is sent 4 times!

> Lorem ipsum dolor sit amet, consectetur adipiscing elit. Integer nec odio. Praesent libero. Sed cursus ante dapibus diam. Sed nisi. Nulla quis sem at nibh el
> 
> Lorem ipsum dolor sit amet, consectetur adipiscing elit. Integer nec odio. Praesent libero. Sed cursus ante dapibus diam. Sed nisi. Nulla quis sem at nibh el
> 
> Lorem ipsum dolor sit amet, consectetur adipiscing elit. Integer nec odio. Praesent libero. Sed cursus ante dapibus diam. Sed nisi. Nulla quis sem at nibh el
> 
> Lorem ipsum dolor sit amet, consectetur adipiscing elit. Integer nec odio. Praesent libero. Sed cursus ante dapibus diam. Sed nisi. Nulla quis sem at nibh el

tfmontague, 2016-05-08:
Tested this with `subscribe` and `notify` as well. Get the same behavior.

randdusing, 2016-05-08:
What's the other device running? Does the error message happen at a certain string length, or all the time?

tfmontague, 2016-05-09:
_What's the other device running?_
iOS 8, iPhone 4S

_Does the error message happen at a certain string length, or all the time?_
There's no error message. The data is sent 4 times when the value is over 157 bytes.

closed 2016-06-20 by tfmontague