randdusing/cordova-plugin-bluetoothle #283: Support Scanning on Windows 10

timfish, 2016-05-08:
I am aware that the Windows 10 Bluetooth API is incomplete and different to every other platform, but can we not support scan discovery using [`BluetoothLEAdvertisementWatcher`](https://msdn.microsoft.com/en-us/library/windows.devices.bluetooth.advertisement.bluetoothleadvertisementwatcher.aspx?cs-save-lang=1&cs-lang=javascript#code-snippet-1)?

This [Stackoverflow answer](http://stackoverflow.com/a/31506096/106623) appears to show it'll work and pass back all the required data...

randdusing, 2016-05-08:
I don't have a Windows development environment anymore, but someone else may be able to add it.

timfish, 2016-05-09:
No worries, I'll do a pull request when I have to implement this myself.