randdusing/cordova-plugin-bluetoothle #284: Windows scan issues, and connect using mac address

cgbrucezjy, 2016-05-09:
So I've been playing around with the windows js code for the ble plugin here's what I figured and have question with.

first I will always get 0 devices find if I use the provided selector
var selector = "System.Devices.InterfaceClassGuid:=\"{6E3BB679-4372-40C8-9EAA-4509DF260CD8}\" AND System.Devices.InterfaceEnabled:=System.StructuredQueryType.Boolean#True";
I don't know where did you get the uuid for ble device.
If I make the selector to "" empty string, I can scan device and see device infos, but I can not connect to it.
I get stuck at deviceInfo.findAllAsync(selector, null).then(function (services) where the selector this time is containerID at service method.

Any idea why this is happening?

Also, Is there anyway I can connect to a hidden bluetooth printer with just the mac address without pairing first?

Thank you so much for your help

randdusing, 2016-07-09:
Sorry for the late reply, but I'm unable to support the Windows version of the plugin.

vladimir-kotikov, 2016-07-20:
I suppose this UUID is for Bluetooth devices that provide `GattDeviceService`. The following code returns similar selector
```js
Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService.getDeviceSelectorFromUuid(
            Windows.Devices.Bluetooth.GenericAttributeProfile.GattServiceUuids.genericAttribute)
```

However, IMO `initialize` method should enumerate available Bluetooth _host controllers_ for the current device rather that Bluetooth _devices around_

randdusing, 2016-07-23:
@cgbrucezjy I merged Vlad's PR #313. Does this help with your issue?