randdusing/cordova-plugin-bluetoothle #288: service not found in characteristic function IOS

Sandervandenberg01, 2016-05-31:
Good day,

I'm getting "Service not found" in the characteristics error callback. The service is found in services() but returns an error on characteristics(). I first tried the regular route threw discover() and connect() but received "Service not found" in the connect() error callback. I'm not having problems with 128 bit UUID's. Iphone 4s (9.3.2). Same problems occur on Iphone 5.

Here's my code:

`this.services = function(checkLock, signout, succes)
{
	bluetoothle.services(
		function(services)
		{
			console.log('[services] succes', services);
			_this.characteristics(checkLock, signout, succes, services.serviceUuids[0]);
		},
		function(error)
		{
			console.log('[services] error', error);
		},{
			address: _this.device,
			services: ["180f"]
		}
	);
};`

[services] succes
Object
address: "8E7765B9-2617-386B-62D6-F2044451C2D4"
name: "00F7D4"
serviceUuids: ["180f", "590b1001-f63e-4175-8ff7-81338429982c", "ce281101-8cb6-4b66-8c4b-2e828d3f067c"] (3)
status: "services"

===============================================

`this.characteristics = function(checkLock, signout, succes, service)
{
	console.log('finding characteristics for service', service);

	bluetoothle.characteristics(
		function(characteristics)
		{
			console.log('[characteristics] succes', characteristics);
			_this.readNearByBike(checkLock, signout, succes)
		},
		function(error)
		{
			console.log('[characteristics] error', error);
		},{
			address: _this.device,
			service: service,
			characteristics: ["2A19"]
		}
	);
};`

[characteristics] error
Object
address: "8E7765B9-2617-386B-62D6-F2044451C2D4"
error: "service"
message: "Service not found"
name: "00F7D4"

Any ideas? If you need more information please tell me.

randdusing, 2016-05-31:
I don't think you're passing the correct service uuid to
your characteristics function. That code will most likely lead it to be
undefined.

Sandervandenberg01, 2016-05-31:
Thank you for your reply.

When I console.log() variable "service" in the error callback I get 180f. The same is outputted before calling bluetoothle.characteristics(). So variable "service" is definitely 180f.

kind regards

randdusing, 2016-05-31:
What version are you using? services.serviceUuids[0] wouldn't work as of v3+. You could add some debug statements to the iOS code and see what values are actually being passed through.

Sandervandenberg01, 2016-06-01:
I'm using 4.0.0. Not using the results of services() and manually filling the parameter with "180f" doesn't work either.

I'm going to debug in xcode.

Sandervandenberg01, 2016-06-01:
I logged some variables:

In BluetoothLePlugin.m 

**function characteristics**
NSDictionary* obj = [self getArgsObject:command.arguments];
{
    address = "ED449EB6-87E5-11B9-5EF8-C021A6641853";
    characteristics =     (
        2A19
    );
    service =     (
        180f
    );
}

**function getService:**
peripheral.services has 3 services and the first one is the battery service:
`<CBService: 0x145d6100, isPrimary = YES, UUID = Battery>`
Shouldn't UUID be 180f?

NSString* uuidString = [obj valueForKey:keyServiceUuid]; 
uuidString is null

Sandervandenberg01, 2016-06-01:
Got it working.

Most probably caused by cordova version 5.4.1
Or possibly cordova ios platform version 3.9.2
Or possibly a combination of the two above.

I fixed it by copying all files to another xcode project which was created with different versions, built the application, and it works.

randdusing, 2016-06-01:
Hmm, keyServiceUuid isn't in 4.0.0 either. Maybe the old project was holding onto an older version of the plugin

Sandervandenberg01, 2016-06-02:
Okay well, strange. Nice to know for future issues.

Thanks

closed 2016-06-02 by Sandervandenberg01