randdusing/cordova-plugin-bluetoothle #289: no device found during scan

giorgioarcella, 2016-06-01:
Hi,
i use startScan, on Android (i checked permission and location enabled), but even if there are many devices recognized during operative system's scan, in my app the callback is never launched after scan..i mean, in "status" there is no "scanResult", only "scanStarted" at the beginning..what am i doing wrong?
```
bluetoothle.startScan(function(result){
            myUtils.log(result);

            if (result.status === "scanStarted") {

                myUtils.log("Scanning for devices (will continue to scan until you select a device)...", "status");
            }
            else if (result.status === "scanResult") {
               //never goes here even if there are devices with bluetooth active
                if (!foundDevices.some(function (device) {
                    return device.address === result.address;
                })) {
                    myUtils.log('FOUND DEVICE:');
                    myUtils.log(result);
                    self.foundDevices.push(result);
                }
            }
        }, function(err){
            myUtils.log(err, 'error');
        }, { services: [] });`

giorgioarcella, 2016-06-01:
solved reading [https://github.com/randdusing/cordova-plugin-bluetoothle/issues/238](url) :D

giorgioarcella, 2016-06-01:
no, really not solved at all, since with same settings now i can't see no more neither the raspberryPI nor other phones with bluetooth enabled...these are all unpaired devices..here is my code:
`startScanning: function(callback){
        var self = this;
        
        if (window.cordova.platformId === "windows") {
            bluetoothle.retrieveConnected(
                self.scanSuccess, 
                self.scanError, 
                {}
            );
        }else{
            bluetoothle.startScan(
                self.scanSuccess, 
                self.scanError, 
                { 
                    services: [],
                    "allowDuplicates": true,
                    "scanMode": bluetoothle.SCAN_MODE_LOW_LATENCY,
                    "matchMode": bluetoothle.MATCH_MODE_AGGRESSIVE,
                    "matchNum": bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
                    "callbackType": bluetoothle.CALLBACK_TYPE_ALL_MATCHES,
                }
            );
        }
        var tentativi = 5;
        var scaninterval = setInterval(function(){
            if(self.foundDevices.length > 0){
                clearInterval(scaninterval);
                self.stopScanning(function(status){
                    //Callback
                    callback(self.foundDevices);
                });
            }
            if(tentativi == 0){
                self.stopScanning(function(status){
                    //Callback
                    myUtils.log('Non ho trovato nulla');
                    callback(self.foundDevices);
                });
            }
            tentativi--;
        }, 5000);
    },`

randdusing, 2016-06-01:
Are the devices advertising in BLE mode?

wifisher, 2016-06-01:
I have the same problem with devices not being found during the scan on Android.

For me, the issue seems to be related to new app installs. I'll explain.

I took an app that I originally developed using version 2.7.1 of the bluetoothle plugin and used it as the base of a new app. In the process I changed the app's id in config.xml and updated all the plugins, etc. With the new app, it never seems to find any of my BLE devices.

So I went back to the original app and it worked as is. I proceeded to update its plugins, etc. (now on bluetoothle v4.0.0). Built and installed and it worked fine.

On a whim, I made two changes to the config.xml file: I added a "2" to the widget id and to the name. I built and ran it and this "new" app (which is the same code as the old app) does not find any BLE devices. Removing the "2"s and going back to the original id and name results in an app that again does find my BLE devices.

So, there appears to be something about new app installs on Android that is resulting the scan not finding devices.

Wayne

wifisher, 2016-06-01:
I now am able to find BLE devices with the new app. The fix is more of a work-around then a fix.

To make it work, I went into Android's app manager and enabled the "location" permission for the app. This single change is sufficient to make it work.

I don't know why the location permission is needed. The app never asked for this permission either.

This is running on a Nexus 7 tablet running Android 6.0.1.

Wayne

randdusing, 2016-06-01:
@wifisher check out the startScan section of the read me, it explains the Android 6.0 permission issues

wifisher, 2016-06-01:
I see that now, thanks. I guess that I was focusing too much on the change from serviceUuids to services and missed the Android details.

But here's a question, if these permissions are required in order for the scan to work, why doesn't the plugin check for them and request them itself instead of pushing that logic into every app? Or maybe return an error or warning if it doesn't have the appropriate permissions?

For implementing this, can I call hasPermission(), requestPermission(), isLocationEnabled(), and requestLocation() from any Android version (at least the hasPermission() and isLocationEnabled() functions) and maybe even iOS? Or do I have to check versions and platforms and only call them where strictly required.

Thanks,
Wayne

randdusing, 2016-06-02:
Main reason was to allow the user to have finer control and be able to show
permissions explanations a little easier.

Do you use Angular? I could modify that pretty easily to allow automatic
requesting of permissions.

If you call them on inappropriate platforms or versions, you'll receive an
unsupported error on iOS or success on Android versions lower than 6.

wifisher, 2016-06-02:
My personal preference for cross platform libraries, etc is always to do the most common action and to work as similarly as possible across the platforms. If the developer wants more control, then the automatic permissions could be disabled by an option in the params object.

I don't use Angular. It is yet one more framework that I haven't had time to look into.

For my purpose at hand, the manual toggle of the permissions is sufficient. Longer term I can either add a couple more state in my scan state machine, or just chain the callbacks.

Thanks,
Wayne

PS. Thanks for the quick response and for maintaining this plugin. I've only used it in some test/debug apps so far, but it has been quite useful. Much appreciated.

randdusing, 2016-06-02:
I'll add that to my list, but PRs are welcome. I'd probably make it not automatic by default unless the major version number is increased.

giorgioarcella, 2016-06-06:
@randdusing the device is a raspberry Pi, and found that it's recognized only when i type 

> sudo hciconfig hci leadv 3

on it, don't understand why. 
I thought an iphone with Bluetooth active would also be detected by my app on android, but it doesn't detect any device but that raspberry pi, after the command..think there is something about bluetooth that i simply don't know

randdusing, 2016-06-06:
The iPhone must be advertising as a BLE device, or it won't be discovered. It won't be advertising by default. An app needs to start advertising.

giorgioarcella, 2016-06-07:
ok, now it's clear, thank you!

giorgioarcella, 2016-06-08:
Another thing, my raspberry sends there infos:
```
[
  {
    "address": "B8:27:EB:9F:5A:43",
    "name": null,
    "rssi": -61,
    "advertisement": "AgECAgoMAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=",
    "status": "scanResult"
  }
]
```

Is there a way to make "name" not null?

randdusing, 2016-06-08:
Is the name null during the scan as well? The Raspberry Pi code may just not ever be setting a name.