randdusing/cordova-plugin-bluetoothle #290: Always getting isEnabled:false from isEnabled

Sandervandenberg01, 2016-06-02:
Good day,

bluetoothle.isEnabled always returns false;

Any ideas?

	this.isBluetoothEnabled = function(callback)
	{
		bluetoothle.isEnabled(
			function(adapter){
				console.log(adapter.isEnabled);
			}
		);
	};

Sandervandenberg01, 2016-06-02:
As a workaround I'll be using bluetoothle.initialize() it does return the correct adapter status.

randdusing, 2016-06-03:
That's very strange. Can you reproduce it on any other devices? If you run the Ionic based example, does the same issue occur?

Sandervandenberg01, 2016-06-03:
Where can I find the ionic based example?

randdusing, 2016-06-03:
It's part of the Angular wrapper:
https://github.com/randdusing/ng-cordova-bluetoothle

davidofwatkins, 2017-04-17:
I'm also having this problem, both on a Pixel (Android 7.2) and Galaxy S7 (Android 7.0):

```js
bluetoothle.isEnabled(function(response) {
    console.log(response.isEnabled); // always false
});
```

randdusing, 2017-04-18:
Does the plugin work other than isEnabled returning false?

davidofwatkins, 2017-04-19:
Oops, my bad. I didn't realize `initialize()` has to be called before `isEnabled()`. As long as `initialize()` is called first, `isEnabled()` does indeed work as expected. Sorry for the false alarm!

That said, I wonder if `isEnabled()` should return `null` or throw an error if `initialize()` hasn't been called yet. It wasn't immediately obvious to me that `initialize()` has to be called even to check the status of the BT adapter.

randdusing, 2017-04-20:
Yah it probably should return an error to be more consistent.

lonewolf73, 2018-11-07:
Hello everyone! 

Sorry to "re-open" this thread but that's what I am looking for after keeping fighting with it for two days...
Anyway I am trying to do a simple sample to work with bluetooth adapter on my mobile phone. I put on home.ts in my sample:

`import { Component } from '@angular/core';
import { NavController } from 'ionic-angular';
import { BluetoothLE } from '@ionic-native/bluetooth-le';

@Component({
  selector: 'page-home',
  templateUrl: 'home.html'
})
export class HomePage {
  public ble_status : string;

  constructor(public navCtrl: NavController, private BLE : BluetoothLE) {
    if (typeof BLE != 'undefined' && BLE) {
      BLE.initialize();
      BLE.isEnabled().then(result => { result ? this.ble_status = "Bluetooth is enabled" : "BLuetooth is disabled"; });
    } else { this.ble_status = "undefined variable" }
  }
  
}`

but the label on the home.html always report "bluetooth is enabled" even the adapter is disabled and so it should report me that's disabled.... I supposed maybe that isEnabled() promise would work as opposite to me so I tried also to enable manually the bluetooth on my mobile phone but it reports still "bluetooth is enabled"..... shall I have missed something then? 

btw, plugin version is cordova-plugin-bluetoothle 4.4.4 "Bluetooth LE"

Thanks to all
Cheers
Luigi

lonewolf73, 2018-11-07:
nevermind.... figured it out right now.... had to put 

`(result.isEnabled ?  "Bluetooth is enabled" : "BLuetooth is disabled")`

instead of 

`(result ?  "Bluetooth is enabled" : "BLuetooth is disabled")`

Thanks anyway!
Cheers!
Luigi