randdusing/cordova-plugin-bluetoothle #293: "Service not found" when writing to existing service.

norenma, 2016-06-09:
I have a problem connecting two iPads to each other. I have one set up as server. When I search and find the device I get the status: 

`{"status":"scanResult","advertisement":{"solicitedServiceUuids":[],"overflowServiceUuids":[],"localName":"Game!!","isConnectable":true,"serviceData":{},"serviceUuids":["1234"]},"rssi":-30,"name":"Game!!","address":"4376FB66-ACCF-8744-8A10-209E03223D40"}`

After connecting I get this status:
`{"name":"Game!!","address":"13B7A54D-FE82-D4C9-3EEB-2BB52620D887","status":"connected"}`
So all good and well. 

Clearly this device has a service called "1234", which I have started. However, when I run anything that refer to the service "1234" on device "1C93C29F-7D05-5EF7-ED4B-1AE9A97EB16D" I get error that no such service exist. 
`
{"message":"Service not found","name":"iPad","error":"service","address":"00A78C7B-A360-8227-D68C-5C2478E3E4D8"}`

I also find this a little bit curious that this shows the senders display-name, rather than the servers. Could that be part of the problem?

The params are logged before I use _bluetoothle.write_:
`{"value":"U2VuZCBvdmVyIGJsdWV0b290aA==","service":"1234","characteristic":"ABCD","type":"noResponse","address":"00A78C7B-A360-8227-D68C-5C2478E3E4D8"}`

The function used to write: 

```
  public write(address){
    let string = "Send over bluetooth";
    let bytes = bluetoothle.stringToBytes(string);
    let encodedString = bluetoothle.bytesToEncodedString(bytes);

    let params = {"value": encodedString,"service": "1234","characteristic":"ABCD","type":"noResponse","address":address};
    let writeSuccess = (status) => {
      console.log("wrote!!");
      console.log(status);
    };
    let writeError = (status) => {
      console.log("error when writing");
      console.log(status);
    };
    console.log("Params: ");
    console.log(params);
    bluetoothle.write(writeSuccess, writeError, params);
  }
```

randdusing, 2016-06-10:
Just to be sure, the device's services and characteristics are being discovered via the discover() or services()/characteristics() functions right?

I've noticed the name thing too. It seems like iOS likes to rename your device to iPad or iPhone occasionally.

norenma, 2016-06-10:
No, I though that since I started the service myself and knew the uuid of it I could just hard-code it to see if it worked. Now when I discovered them it works.

Thanks a lot for your awesome work!

randdusing, 2016-06-10:
Glad it's working.

closed 2016-06-10 by randdusing