randdusing/cordova-plugin-bluetoothle #296: bluetoothle not found error

ka347, 2016-06-24:


randdusing, 2016-06-24:
Are you calling it before the deviceready event has fired?

ka347, 2016-06-24:
no

randdusing, 2016-06-24:
Post the relevant code and I'll see if I can figure out the issue

ka347, 2016-06-24:
i am new to this and when user clicks myfun2() will be called 
 var params = {
        "request": true,
        "statusReceiver": false,
        "restoreKey": "bluetoothleplugin"
    }
    bluetoothle.initialize(function (result) {
        alert("heyy");
    }, [params]);

randdusing, 2016-06-25:
That looks fine, but when is the code fired?

ka347, 2016-06-25:
when click on the button scan for ble the code gets executed

randdusing, 2016-06-25:
If you can post all your code, that would help a lot.

ka347, 2016-06-26:
hey i uninsatlled and reinstalled its working, but now when i scan i am getting error as scannning already in progress

randdusing, 2016-06-27:
I can't really offer much help without taking a look at the code.

closed 2016-07-04 by ka347