randdusing/cordova-plugin-bluetoothle #298: Max amount of packages per second

norenma, 2016-06-27:
Hi! 

I would first like to thank you for your awesome work!

I use your plugin to develop a multiplayer iPad game. I have 4 connected iPads and one acting as server. Right now each iPad can send 8 packages per second to the server and the server notifies all the other iPads about the change. So right now the server sometimes need to send 32 packages per second. I would however need to send more than this, but when I try to increase this the packages seem to fall behind and are sent much later. Is there a limit to how often a service can notify of a change? Is it dependent on the number of connected clients? 

Thanks in advance, 
Markus

randdusing, 2016-06-27:
How much data is being sent per package? I did some testing a while back with my sample app to see how quickly I could notify on a single connection, but I don't have the results recorded anywhere unfortunately.

norenma, 2016-06-29:
It seems now that I am able to send packages each 30 ms. Apparently the problem was somewhere else in my code. As this works it seems like the peripherals has a connection interval of 30 ms, which is from what I understand it the lowest possible on iOS, so no complaints there. 

Anyways great job and sorry for the inaccurate issue!

randdusing, 2016-06-30:
No worries, I'll keep this open to remind myself to document this. It seems like you can probably send a notification every X ms where X is the connection interval. I'll have to see how this scales if you multiple devices are subscribed to the same server.