randdusing/cordova-plugin-bluetoothle #303: Slower write when restarting app/cleanup problem

norenma, 2016-06-30:
Hi, 

When I exit my iOS-app though force quit in the multitasking-view, and then restart it, create a service with a characteristic that I'm subscribing to on the clients, the write speed and/or notify speed seem to go down. 

Is there anything that is not destroyed when I force quit the app? When I start the app on all devices from Xcode I don't have this problem. 

The service and characteristic has the same name on each run. I have tried to call unsubscribe and close on the client devices, and removeAllServices on the server.

Thanks in advance, 
Markus

randdusing, 2016-07-06:
How often does the 2-3 second lag occur? I haven't noticed the lag, but usually I didn't send messages that quick nor to more than one subscriber.

norenma, 2016-07-15:
I am trying to understand better when this occur. It seems to me that very often this happens when I have used bluetooth a intensively for a while, but not necessarily when the app has been opened for a long time; sometimes when I restart the app and use it intensively several times it can start off very slow. As if either the bluetooth unit is  exhausted somehow, or is it possible that there is some effect that I can have on bluetooth that persists after the app is terminated though the multitasking view? 

Anyways, I know it's not a lot to go on, just though I'd let you know and see if you have any ideas. 

Thanks again for your awesome work. 
/Markus