randdusing/cordova-plugin-bluetoothle #304: startAdvertising

ka347, 2016-07-04:
hey, when we start advertising from phone what will be id that is seen on other phone when its scanning

norenma, 2016-07-04:
On iOS: If you are talking about the address, this is a device-specific and generated, which means it is not equal to the mac-address of the devices, and it will not be the same from session to session, and not the same on different devices that finds the same device. 

On Android: I am fairly certain that this is the mac address.

ka347, 2016-07-04:
function myfun(){ 
bluetoothle.startAdvertising(function (result) {
         alert("success advertising" + JSON.stringify(result) + "data is ");
     }, function (error) {
         alert(JSON.stringify(error));
     }, {
         uuid: 'CDB7950D-73F1-4D4D-8E47-C090502DBD63',
         services: ["1234"],
         service: "1234",
         name: "Hello World",
         mode: "highLatency",
         connectable: false,
         timeout: 10000,

     });
}
is this correct??

ka347, 2016-07-04:
when we scan fom rpi will it show this uuid

norenma, 2016-07-04:
You do not need to specify a uuid when you start advertising, it will be generated. 

It seems somewhat correct. I don't know about the extra parameters that you added, you only need a name, a service(or array of serviced if you use iOS). 

Did you successfully call initializePeripheral and addService before calling startAdvertising?

ka347, 2016-07-05:
hey , i am getting an error attempt to invoke virtual method 'boolean android.bluetooth.BluetoothGattServer.addService(android.bluetooth.BluetoothGattService)' on a null object reference

ka347, 2016-07-05:
i got that i used initialize() not initiliazePeripheral.Now, my device is saying advertising in phone 1has started but when i scan in other phone 2 it is only showing the phone 1 name.its not showing anything other than phone 1 as the availbale device, i mean it should show the uuid of advertising data.
thanks in advance.

norenma, 2016-07-05:
How do you display the information? The uuid is in a variable called 'address' in the object returned when scanning.

ka347, 2016-07-05:
in android its showing mac address not the uuid

randdusing, 2016-07-05:
Android doesn't encrypt the address into a UUID like iOS. Are you looking for the service UUID?

ka347, 2016-07-06:
but when i scan for ble from raspberry pi its showing the device id and uuid of phone1. but i dont know what is the uuid its showing (bluetooth uuid of my phone or some generated uuid)

randdusing, 2016-07-07:
Not sure if I follow. I'm not sure what format the raspberry pi returns scan results in.

ka347, 2016-07-07:
7e:23:b7:7b:26:12,010204094d692006ff010045a9dd020a,63747,820,18,-71
id,uuid,major,minor,rssi
this how rpi is displaying

randdusing, 2016-07-07:
The first part definitely looks like a MAC address, but I'm not sure what the UUID represents.

bharath26, 2016-07-08:
hey norenma, you told

"You do not need to specify a uuid when you start advertising, it will be generated".
can you tell me on what value sthe uuid gets generated

ka347, 2016-07-08:
hey norenma, you told

"You do not need to specify a uuid when you start advertising, it will be generated".
can you tell me on what value sthe uuid gets generated

randdusing, 2016-07-08:
The UUID is generated by iOS and you have no control over it.

ka347, 2016-07-12:
okay,thanks.

closed 2016-07-12 by ka347

ka347, 2016-07-12:
hey, i have one more doubt.
cant i pass a variable in startAdvertising parameters.i mean when i pass a variable in manufacturerSpecificData and when i scan and read the data it is showing me undefined

bytes = bluetoothle.stringToBytes(profile_id);
encodedString = bluetoothle.bytesToEncodedString(bytes);

 bluetoothle.startAdvertising(function (result) {
        alert(bytes+"   "+encodedString);
        alert("success advertising" + JSON.stringify(result));
    }, function (error) {
        alert("error" + JSON.stringify(error));
    }, {
        services: ["1235"],
        service: "1235",
        mode: "lowLatency",
        timeout: 500,
        powerLevel: "low",
        manufacturerId: 0x004C,
       //manufacturerSpecificData:  bluetoothle.bytesToEncodedString(profile_id),
       //manufacturerSpecificData: 'OTg0MTEyMzQ1'
       manufacturerSpecificData: encodedString

    });
}