randdusing/cordova-plugin-bluetoothle #305: Android: startScan error

tomaszpupiec, 2016-07-05:
Good day,

My environment:

Phone: Samsung Galaxy S6
Phonegap: 
	phonegap plugin list
	cordova-plugin-bluetoothle 4.0.0 "Bluetooth LE"
	cordova-plugin-compat 1.0.0 "Compat"
	cordova-plugin-whitelist 1.2.2 "Whitelist"

	phonegap platform
	Installed platforms: android 5.2.0	
	
Code:	

var scanSuccess = function (device) {
			console.log(device)
		};

var scanFailure = function (error) {
			console.log(error);
		};

bluetoothle.initialize(function () {
  bluetoothle.startScan(scanSuccess, scanFailure);
});
	
When I run above code, scanFailure is calling with error "Attempt to invoke virtual method 'int org.json.JSONObject.optInt(java.lang.String, int)' on a null object reference".
Any ideas what's a problem ?

randdusing, 2016-07-05:
Hi Tomasz, try adding an empty object parameter after scanFailure in the startScan function. The Android code needs better bounds checking.

tomaszpupiec, 2016-07-06:
Works. Thanks !

closed 2016-07-06 by tomaszpupiec

randdusing, 2016-07-09:
This is now fixed in master, and I'll be releasing 4.1.0 sometime today