randdusing/cordova-plugin-bluetoothle #308: P2P Android and/or iOS

oktau, 2016-07-13:
Hello,

First of all, thank you for the work done on this plugin.
Can you tell me if is it possible to establish a two ways communication (Android/Android, iOS/iOS or Android/iOS) with this plugin? If yes, is there an example available?

Thank you for your answers.

Best regards.

randdusing, 2016-07-14:
The Angular wrapper has an example of one way communication that you can check out and then reverse. I don't remember if I tried two way communication while testing, but I think it should work.

oktau, 2016-07-22:
Hello,

Thank you for your answers. I have done a test with the angular example. Indeed, the two ways communication seems to work properly.

Thanks again.

closed 2016-07-22 by oktau