randdusing/cordova-plugin-bluetoothle #309: startScan() method returns nothing on IOS

fabLouis, 2016-07-19:
Hello,

When I call _startScan()_ method (with {} for params), the callbacks success or error are not raised only on iOS, on Android it's work well. Why?

My environment:
```
$ cordova -v
6.1.1
$ ionic platform
Installed platforms:
  android 5.1.1
  ios 4.1.1
$ ionic plugin list
com.verso.cordova.clipboard 0.1.0 "Clipboard"
cordova-plugin-bluetoothle 4.2.0 "Bluetooth LE"
cordova-plugin-camera 2.2.0 "Camera"
cordova-plugin-compat 1.0.0 "Compat"
cordova-plugin-console 1.0.3 "Console"
cordova-plugin-device 1.1.2 "Device"
cordova-plugin-email-composer 0.8.3 "EmailComposer"
cordova-plugin-file 4.2.0 "File"
cordova-plugin-file-transfer 1.5.1 "File Transfer"
cordova-plugin-google-analytics 0.8.1 "Google Universal Analytics Plugin"
cordova-plugin-inappbrowser 1.4.0 "InAppBrowser"
cordova-plugin-splashscreen 4.0.0 "Splashscreen"
cordova-plugin-statusbar 2.1.3 "StatusBar"
cordova-plugin-whitelist 1.2.2 "Whitelist"
ionic-plugin-keyboard 2.2.0 "Keyboard"
phonegap-plugin-push 1.7.0 "PushPlugin"
```

The (basic) code:
```
this.bleComponent.startScan(
    function(results) {
        console.debug('startScan > success: ' + JSON.stringify(results));
    }, 
    function(err) {
        console.error('startScan > error: ' + JSON.stringify(err));
    },
    {}
);
...
setTimeout(function () {
    this.stopScan();
}, this.scanningDelay);
```

Tested on iPhone 5c (iOS 9.3.2).

Thanks for your help

randdusing, 2016-07-20:
Can you post the source for bleComponent?

fabLouis, 2016-07-20:
I think it's not important because the bleComponent is a wrapper class for the opportunity to easily change (in case) the BLE plugin.

For the exemple for the bleComponent.startScan() method:
```
public startScan(successCallback, errorCallback, params) {
    this.bluetoothle.startScan(successCallback, errorCallback, params);
}
```

randdusing, 2016-07-23:
Maybe try removing and readding the plugin. Also the ```this.``` in ```this.bluetoothle```could potentially be causing an issue.

fabLouis, 2016-07-26:
Same result Rand.
About the `this.bluetoothle`, the `this.` is here because the language is TypeScript.

randdusing, 2016-07-28:
Do you have Ionic installed? You could try using the example in https://github.com/randdusing/ng-cordova-bluetoothle. If even that doesn't work, it may be device related.