randdusing/cordova-plugin-bluetoothle #310: Speed up writing large amounts of data?

dibari, 2016-07-19:
First, thanks for all your work on this great plugin.  We're sending a firmware update to a device, which consists of about 10k lines of 18 bit data packages.  We have a native iOS implementation for the firmware updating, and it takes about 2.5 minutes.  With the plugin through cordova, it's about 8 minutes.  It's able to send each packet and receive a response in about 40-70ms.

I'm assuming not, but Is there anything I could try to speed up sending all that data?

randdusing, 2016-07-20:
How often are you calling write, and is it in parallel or sequential? The plugin to Javascript bridge does add some overhead. Are you aren't using write without response on the native app and normal writes with the plugin? And how many bytes are you writing before waiting for an acknowledge from the device?

dibari, 2016-07-20:
I'm writing in sequence.  More importantly, it looks like the native app is writing without response.  In the cordova app, I'm writing 18 bytes and waiting for a response.

Right now, I'm subscribing to the appropriate service and characteristic.  Should I be using addService() instead, and setting writeWithoutResponse to FALSE?  Am I thinking of the addService() function properly?

randdusing, 2016-07-23:
```addService()``` is for devices acting as a server. You just need to set the ```type``` parameter in ```write()```to ```noResponse```. See [doc](https://github.com/randdusing/cordova-plugin-bluetoothle#write)

Writing without response should bring the performance much closer to a native app, but you'll still have the performance decrease from the Javascript <> Native bridge.

dibari, 2016-08-05:
Thanks for the reply.  I was able to set noResponse, and performance seems about 80/85% of what I'm getting in native.  Keep up the great work!

closed 2016-08-05 by dibari