randdusing/cordova-plugin-bluetoothle #312: No Popup on initialize()

LucasChapuis, 2016-07-21:
I've got this 

<pre> new Promise(function (resolve, reject) {

            bluetoothle.initialize(resolve, reject,
                { request: true, statusReceiver: false });

        }).then(initializeSuccess, handleError); </pre>

Is that normal to not have any prompt to ask to turn on bluetooth while the **request** attribute is set to *true* when the Bluetooth is disabled ?

Thanks in advance !

vladimir-kotikov, 2016-07-22:
The sample is incorrect - `initialize` method does not accept error callback. This
```js
bluetoothle.initialize(resolve, { request: true, statusReceiver: false });
```
should work.

randdusing, 2016-07-23:
Thanks @vladimir-kotikov ```initialize()```has gone through some changes and the documentation has been out of sync. @LucasChapuis If this doesn't resolve your issue, please reopen.

closed 2016-07-23 by randdusing