randdusing/cordova-plugin-bluetoothle #316: Location always shown as Enabled

LucasChapuis, 2016-07-26:
I've got this code : 
<pre>
$("#Test-Activ-Location").click(function () {

    new Promise(function (resolve, reject) {

        bluetoothle.isLocationEnabled(resolve, reject);

    }).then(isLocationEnabledSuccess, handleError);

    //bluetoothle.requestLocation(requestLocationSuccess, requestLocationError);

});
</pre>
<pre>
function isLocationEnabledSuccess(status)
{
    if(status.isLocationEnabled === true)
    {
        log("La localisation est activée.", "succes", "Log-Scan-Bluetooth");
    }
    else
    {
        new Promise(function (resolve, reject) {

            bluetoothle.requestLocation(resolve, reject);

        }).then(requestLocationSuccess, handleError);
    }
}
</pre>
<pre>
function requestLocationSuccess(status) {

    if(status.requestLocation === true)
    {
        log("La localisation est activée.", "succes", "Log-Scan-Bluetooth");
    }
    else if(status.requestLocation === false)
    {
        log("La localisation n'est pas activée", "warning", "Log-Scan-Bluetooth");
    }
}
</pre>
<br>
<p>So if the location is enabled it should log "it's activated", otherwise it should open the prompt and ask to enable/disable location, then log  "it's not activated" if we didnt check it or "it's actived" if we did</p>
<p>But the main problem here is that it's always telling me it's activated, even when it's not, so i tried different combinings.</p>
<p>
<ul>
<li>Activated --> launch application --> Test --> log : activated</li>
<li>Not activated --> launch application -->Test --> log : activated</li>
<li>Activated --> launch application -->manually desactivate --> Test -->log : activated</li>
</ul>
</p>
<p>If you've got any idea why it's always telling me it's enabled, I'm glad :D</p>
<br>
<p>Pictures : </p>
<a target="_blank" href="http://img15.hostingpics.net/pics/874828localtiondisabled.png">Location disabled</a>
<a target="_blank" href="http://img15.hostingpics.net/pics/874828localtiondisabled.png">Code always on "enable"</a>

randdusing, 2016-07-29:
What version of Android are you on? This function is only intended to be use with Android 6.0 because Location being disabled doesn't affect BLE on earlier Android versions

LucasChapuis, 2016-07-29:
I'm on Android 5.0.2 lollipop, didnt see the (Android 6+) on the function. It seems like Marshmallow is asking way more often for permissions.
Btw I thank you for your plugin, it's a great one, easy to use, clear, and with a really nice doc' :)

closed 2016-07-29 by LucasChapuis